EIPStackGroup/OpENer #31: Compliance Test fails for Connection Manager / Unconnected Send

emb4fun, 2015-01-11:
With issue #26 there was an updated implemented for "Unconnected Send" feature.
But this feature breaks the compliance test (CT). For the CT the new feature
"0x52 Unconnected_Send" was enabled too. And in this case the following errors occur:

STC File Error: Connection Manager Unconnected_Send supported and STC file does not indicate CIP routing support

STC indicates that Connection Manager Unconnected_Send service is implemented and test is unable to retrieve entry port. 'One instance of the Port object shall exist for each CIP routable port

It looks that the "CIP routing" and even the "Port" feature is needed for "Unconnected Send"?!

I think the patch from issue #26 should be removed.

azoitl, 2015-01-11:
As the port object and also the routing is quite some effort and for adapter devices with a single port an overhead I would also recomend to remove the patch from isse #26.

CapXilinx, 2015-01-13:
As suggested by azoitl and emb4fun the commit was reverted

commit 195410d1 (2015-01-13): Closes #31 Revert "Merge pull request #26 from RalfBrandt/patch-3"

This reverts commit 8c6ecec63bbb5495a002b7e629206be28b596a0c, reversing
changes made to 5a93582bb4520a5e204dc3ec1a2d578abcca109a.
closed 2015-01-13 by CapXilinx