EIPStackGroup/OpENer #38: EIP Scanner ?

Richard-ITR, 2015-11-03:
Hi all,
I want to bring up this issue again!

Is there any experience to use the OpENer Stack as base for a master/scanner implementation as well. Has anybody started already such a project. Maybe we could collaborate!
 
Beginning of 2016 we want to start an Ethernet/IP device/adapter project.
My experience with other stack porting projects tells me, that it would be better to have source codes on both sides. I would want to create a demo scanner implementation for Windows/PC mainly for testing our scanner implementation (e.g. we would be able to generate wrong telegrams to improve the robustness of our code)
Many thanks

Driffster, 2015-11-03:
I would be interested in such application, although I don't have the background to actually contribute to such a project (aside from testing communication on a controller using this protocol).

CapXilinx, 2015-11-04:
I have not started on such a project, but I plan in the near future would be to implement an explicit messaging client for OpENer. This would be in my opinion a first step to give OpENer full client capabilities.

Richard-ITR, 2015-11-06:
Thanks for your replies!!
@CapXilinx:  I am very new in Ethernet/IP, but I am very experienced in other realtime ethernet protocols (Profinet, Ethercat) and in stack portings.
Could you give a small summary, which functionalities would be needed for full client capabilities?
maybe we could collaborate and work on diffent specific issues.

CapXilinx, 2015-11-16:
The best starting point would probably be ODVA's  Public Specification on Recommended Functionality for EIP Devices, available at http://www.odva.org/Publication-Download

It would be probably also a good idea to first enhance the OpENer to comply with the latest CIP and EtherNet/IP specification, which will be the next thing I will work on.

In the last days I also worked through the code base to update the doxygen comments and to streamline the code to one coding style, which I choose the Google C/C++ style. The changes will be online in the next days.

CapXilinx, 2017-10-11:
As last activity on this issue was over a year ago I will close this issue now.
If full client functionality will be developed by me, it will be a separate project, as although there is overlap between scanner and adapter, the differences in operation are too big in my opinion to bundle both in one project.

closed 2017-10-11 by CapXilinx

alexhjf, 2018-02-28:
Hi Richard-ITR,
Could please tell me how's your EIP scanner implementation project going now?

Richard-ITR, 2018-02-28:
@alexhjf 
Hi, 
there was absolutely no activity in this issue. I do not have enough knowledge about EIP to start a scanner project alone. Maybe this issue will pop up again somewhen, when there is some interest from customers.