EIPStackGroup/OpENer #48: Overhaul Forward Open / Forward Close

CapXilinx, 2015-12-01:
Due to changes in the CIP sepcification these services needs additional functionality, which is not easily achievable with only modifying the existing code. A new implementation is needed

liftoff-sr, 2016-03-02:
Can you provide the CIP specification version number at which these changes were introduced?

CapXilinx, 2016-03-02:
Unfortunately I cannot. As I am no member of the ODVA I do not have access to every specification release. Currently I have access to the April 2015 specification, due to the courtesy of the ODVA and Rockwell. Bevor that I only had access to the September(?) 2009 specification.

mmprestine, 2016-03-03:
There is still a problem with the ForwardOpen code / ConnectionObject.  When I connect clogix processor to OpENer via a Generic Device (CIP Explicit?) I get an extended error 295 with the OpENer traces enabled.  Also, should this not be added to the to the ConnectionObject? 	
```
/* pointers to connection handling functions */
	OpenConnectionFunction  open_connection_function;
```

pogojotz, 2016-08-04:
Could you please provide a summary of the necessary changes to be made, if feasible?

CapXilinx, 2016-08-04:
I recently started working on checking and fixing this issue. The original code is a bit confusing, which is why I think some parts have to be done from scratch.
I cannot easily give a list of needed changes as I only have access to the 2009 and the 2015 specs, somewhere in between at least error responses have changed, also several I/O connection establishment scenarios seem to have changed.

liftoff-sr, 2016-10-11:
I spent time on this for CIPster and will have a new implementation of forward open and large forward open out by end of this week.  This was largely an exercise in parsing.   I will push after more testing.  Wanted to let you know in case is might serve as a model for you, but it would have to be converted to C from C++.

https://github.com/liftoff-sr/CIPster

I also plan on starting a CIPster mailing list this week also.

closed 2017-01-08 by CapXilinx

yanzixiang, 2019-08-26:
@mmprestine 
I think you are using lwip,
can you share your source of generic_networkhandler.c.
I thins We are facing the same problem.

commit 0ce6f69f (2017-01-08): Closes #48 Code cleanup