EIPStackGroup/OpENer #52: gcc options change

liftoff-sr, 2016-03-07:
The attached patch changes CMAKE_C_FLAGS in the following way:

1) remove -w which quiets all warning and causes function prototypes to be ineffective.  That basically renders all *.h files almost useless, except for structure declarations.

2) removes -DCMAKE_EXPORT_COMPILE_COMMANDS=ON which is not a compiler option, so should not be passed to gcc.  Nor is it tested in the C source by the preprocessor.

[cmake.patch.txt](https://github.com/EIPStackGroup/OpENer/files/161636/cmake.patch.txt)

liftoff-sr, 2016-03-07:
I would also suggest this option:

     -Werror=implicit-function-declaration

This makes it an error to use a function for which a function declaration has not yet been seen.

This will trigger a couple of missing function prototype errors, that will subsequently need fixing.  But it's worth it.

I can push my whole tree if you want.

closed 2016-03-08 by CapXilinx

CapXilinx, 2016-03-08:
Thank you for your suggestions!
I already implemented them and fixes the missing prototypes.

A probably easier way for you to share your code modifications would be to fork the repository and then to issue a pull request.

commit e6c027f3 (2016-03-08): Closes #52 Implements the additional suggestion of liftof-sr and fixes missing function prototypes