EIPStackGroup/OpENer #53: Wrong message size in Forward Open message

CapXilinx, 2016-03-08:
In the Forward Open message additional 8 bytes are sent. See discussion Issue #51

CapXilinx, 2016-03-08:
After renaming the parameter n to something more meaningful, I realized that I added the moved and written size n again to n, resulting in an extra 8 byte for connected messages in AssembleLinearMessage.
As I currently do not have a PLC ready for testing could someone verify that the patch is working, bevor I push my commit to the repository?

Thank you!
[fix_issue_53.patch.txt](https://github.com/EIPStackGroup/OpENer/files/163107/fix_issue_53.patch.txt)

mmprestine, 2016-03-08:
Yes, I have just found the issue also.

`int FillNextNMessageOctetsWithValueAndMoveToNextPosition(CipOctet value, unsigned int n, CipOctet **message) {

  n **=** FillNextNMessageOctetsWith(value, n, message);** //+= ADDS AN EXTRA 8 BYTES TO THE END OF THE MESSAGE**

  MoveMessageNOctets(n, message);

  return n;
}`

I have tested and it works with fix.

Thanks
Matt

mmprestine, 2016-03-08:
I just reviewed your fix and it is different that what I have found.  I will test your new code.

mmprestine, 2016-03-08:
I have validated that the code fix for issue 53 is successful.

commit 2157270e (2016-03-08): Fixes #53, corrects helper function which added size twice to return value
closed 2016-03-08 by CapXilinx