EIPStackGroup/OpENer #64: Windows port doesn't build 'out of the box'

giajh01, 2016-07-20:
I am attempting to build OpENer on cygwin. 
./setup_windows.bat  works
./make fails with the error shown below.

Any ideas what I'm doing wrong/have overlooked?

The toolset I'm using is as follows
cmake       3.3.2
gcc            5.4.0
make         4.2.1

[  3%] Building C object src/cip/CMakeFiles/CIP.dir/appcontype.c.o
In file included from /c/proj/OpENer.git/OpENer/source/src/cip/cipconnectionmanager.h:10:0,
                 from /c/proj/OpENer.git/OpENer/source/src/cip/appcontype.h:9,
                 from /c/proj/OpENer.git/OpENer/source/src/cip/appcontype.c:9:
/c/proj/OpENer.git/OpENer/source/src/opener_api.h:421:34: warning: ‘struct sockaddr_in’ declared inside parameter list
                                  int *number_of_remaining_bytes, int unicast);
                                  ^
/c/proj/OpENer.git/OpENer/source/src/opener_api.h:421:34: warning: its scope is only this definition or declaration, which is probably not what you want
/c/proj/OpENer.git/OpENer/source/src/opener_api.h:437:36: warning: ‘struct sockaddr_in’ declared inside parameter list
                             struct sockaddr_in *from_address);
                                    ^
/c/proj/OpENer.git/OpENer/source/src/opener_api.h:631:28: warning: ‘struct sockaddr_in’ declared inside parameter list
                     struct sockaddr_in *socket_data);
                            ^
/c/proj/OpENer.git/OpENer/source/src/opener_api.h:644:13: warning: ‘struct sockaddr_in’ declared inside parameter list
             EipUint16 data_length);
             ^
In file included from /c/proj/OpENer.git/OpENer/source/src/cip/appcontype.h:9:0,
                 from /c/proj/OpENer.git/OpENer/source/src/cip/appcontype.c:9:
/c/proj/OpENer.git/OpENer/source/src/cip/cipconnectionmanager.h:202:22: error: field ‘remote_address’ has incomplete type
   struct sockaddr_in remote_address; /* socket address for produce */

CapXilinx, 2016-07-20:
If you are using Cygwin you are supposed to use setup_posix.sh, as Cygwin is providing a POSIX layer for windows. The setup_windows.bat is for Windows native compiling

giajh01, 2016-07-20:
Duh! Thank you for pointing out the error of my ways

In case anyone else makes the same mistake:
On Windows OpENer can be built 
- using cygwin => use bin/setup_posix.sh
- using cmake + Visual Studio 2013 => use bin/win32/setup_windows.bat

closed 2016-07-20 by giajh01