EIPStackGroup/OpENer #65: Additional RegisterSession command on registered encapsulation session results in "Invalid Command" status in the reply

pogojotz, 2016-08-04:
During debugging while receiving a RegisterSession command from a master/scanner I encountered, that the session was really registered, but on master/scanner side it was discarded due to timeout. This resulted in a one-sided registered encapsulation session. So when the master try to start a new session on the same TCP socket it is refused by `HandleReceivedRegisterSessionCommand()` with status code `1` (Invalid Command).
I didn't find anything on the reissuing of the RegisterSession command in the ODVA specifications of 2007 (the only ones I have right now), I how one should react.
My question is now: Couldn't we just reply with a success status and the already registered session handle or is that a violation of the protocol?

Cheers

CapXilinx, 2016-08-04:
Hi.

I will check this in my April 2015 ODVA specification, 

Cheers

azoitl, 2016-08-04:
According to my memory (I currently don't have the specs at hand) it is not allowed to accept a second register session from the same socket. This is also tested by the conformance test tool. I do remember because I had to add quite some code to handle this correctly.  

As I got some doubts I just rechecked my code and the comments. The problem is that you shouldn't use a new register session but just send commands with your old session id. However as also my comment there states the CIP spec at the time of writing this code (app. 4 years ago) was not clear. The only thing that was clear that the register session needs to be rejected.

CapXilinx, 2016-08-04:
In the 2015 Spec it is stated, that if the originator tries to register more than one session on the same socket, error code 0x0001 shall be returned

azoitl, 2016-08-04:
As the spec is not clear about what session id to return in the unsucessfull return message we also add the correct session id. This information a scanner could use to send correct packets. Not sure if this is a good behavior.

pogojotz, 2016-08-05:
Ok, thank you very much for the clarification!

closed 2016-08-05 by pogojotz