EIPStackGroup/OpENer #74: Can I use this library to talk to

TomaszSli, 2016-10-27:
I am new to Ethernet/IP and was curious if this library can talk and get data from a devices like this using Ethernet/IP:
https://www.acromag.com/catalog/1025/ethernet-modbus-profibus-io/ethernet-io-modules/analog-io/busworks-xt-series/xt1240
It would be a Ubuntu system.  I can build and execute library code and sample. 
Not sure how to proceed further as far as adding code to talk to the device; can’t find any examples.
thank you

CapXilinx, 2016-10-27:
Hi!

Short answer, no you can't.
The OpENer is a CIP Adapter device library. If you have access to the CIP and EtherNet/IP specification it would be possible to enhance OpENer to provide the abitlity to communicate to your input module.

See also Issue #38

closed 2016-10-27 by CapXilinx

TomaszSli, 2016-10-28:
Thank you for the response!  Looks like for our application we will switch our devices for Modbus versions where it seems easier to do a query/response type of thing.  
I think (maybe) I could put a hack together to do an explicit type of query with Ethernet/IP as well..but not sure how much effort it would be.
Thanks again

CapXilinx, 2016-10-28:
Hi! I also want to point out the post of pogojotz in Issue #69. There is a EtherNet/IP explorer, which probably can do what you need (https://sourceforge.net/projects/enipexplorer/)

But I agree that Modbus could be simpler :)

As you said, it should also be possible to create an explicit request, but the standard would be to use implicit messages. You would need a Forward Open message for your device, and then just listen to the generated data.
In addition you would need a Forward Close message to shut down the connection at the end.