EIPStackGroup/OpENer #75: Using OpENer to send other imformation ?(In my opnion)

q753q753, 2016-11-30:
In my opnion, OpENer is a master(maybe wrong?), it provides some imformation let client to read ex ip.gateway...
And I used Ethernet/IP browser on windows and really got some imformation.
But I want to ask that is there other client or slave(maybe server?) connecting to OpENer ?
And I want to let OpENer send other imformattion to client like the cpu's temperture, but I don't know how to write that I can send the imformation?

CapXilinx, 2016-11-30:
Hi!

To be honest, I do not understand your question
Why should the OpENer be a master?
Also you are mixing up several communication models (master/slave, client/server), and besides the CIP communication model is based on a consumer/producer model.

For sure you can send CPU temperature data values, but as OpENer is a CIP adapter you first have to request that it sends this value. In addition you have to provide in your application code this data value to be sent.

For tools to access the OpENer please see #38. If this was your question I would also ask you to use the search function first, bevor writing a question issue

q753q753, 2016-11-30:
Thank's for your reply ! 

Maybe I just use OpENer recently so that my experience in using OpENer is a master which waiting another devices to link it.
That's why I'm confused OpENer is a master or slave?

About #38,Sorry, I can't find your meaning "tools" , is this 
[](https://www.odva.org/Publication-Download) ?
and I'm not sure which link is about CPU temperature and so on.

CapXilinx, 2016-11-30:
Sorry, my fault it's issue #69 where pogojotz is mentioning a EtherNet/IP scanner named EtherNet/IP Explorer.

A master is not waiting for a connection, it is initializing it to access a slave.

As said CIP is not using a master/slave model but a consumer/producer model. The OpENer is just waiting for a Forward Open request to open a connection and send data.

q753q753, 2016-11-30:
Yeah, #69's tools is that what I'm using in the beginning question .

So, you mean that OpENer is a consumer/producer , and I must use another to send data to it？

Btw,sorry, I don't really understand "Forward Open".

azoitl, 2016-11-30:
In EtherNet/IP you have to main kind of devices. 

1. adapter devices which offer connection endpoints to which other devices can connect to send data or subscribe to data provided by adapter devices. But adapter devices are not able to establish such connections. OpENer is such an adapter device.
2. scanner devices which are able to connect to adapter devices end establish communication. For this a EIP service which is called "Forward Open" is used. 

In order to get a better understanding of how EIP is working I can recommend the documentation provided by ODVA. For example the CIP overview document is for starters to get an idea what is waht a very good starting point. 

For your use case you could use OpENer es such that you implement code that reads the CPU temp and offer it to a scanner device via a so called assembly object. Then the tools as you used them can connect to your device and subscribe to these values.

q753q753, 2016-11-30:
Thanks to your replies !

And the last thing I want to ask that where can I implement my code to read the CPU temp in OpENer ? 
I have seen a little the OpENer-master folder  that there is no similar example to help me to help me realize it.

azoitl, 2016-11-30:
If you llike in the wind32 or Linux prot there you'll find a sample application. This is the place where you'll typically put your application code. There are three methods which you could use:

1. HandleApplication: this function is called by the stack in regular intervalls
2. AfterAssemblyDataReceived: will be invoked after data has been received for a specific assembly object. the demo code there is showing how you determine which assembly object received data and how you cann access this data. It is just copying the received data to the data to send.
3. BeforeAssemblyDataSend is called by the stack just before data should be send. There you can apply similar means as in AfterAssemblyDataReceived to check if you need to update data and set the data of your assembly oject accordingly to your measurment data.

I hope this helps.

q753q753, 2016-12-01:
I use linux and I find a named "sample_application" folder in the /OpENer-master/bin/posix/src/ports/POSIX/sampl
e_application,but I can't open it.

How can I open it ?

azoitl, 2016-12-01:
Maybe you should look for an sample_application.c in OpENer-master/src/ports/POSIX

q753q753, 2016-12-01:
OK. I tried to run it , but there is some bug that i can't run it.
What  the results should be after running it ?

CapXilinx, 2016-12-01:
What's the problem? OpENer is running at my machine without a problem

q753q753, 2016-12-03:
I use terminal to compile the sample.c , maybe i used it in a wrong way?

So the running result of sample_application.c is directly running OpENer ?

azoitl, 2016-12-03:
the sample_application.c is part of a default build of OpENer. You need to build a full OpENer application using cmake and the generated make files. May I refer you to the read me and the extensive documentation Martin wrote.

q753q753, 2016-12-06:
OK I'll try it , thank you !

CapXilinx, 2016-12-20:
As this topic seems to be resolved, I close it now

closed 2016-12-20 by CapXilinx