EIPStackGroup/OpENer #80: Trying to use appcontype.h in C++ fails due to duplicate const error

simonlui, 2017-01-11:
I'm not sure if this is valid since this is intended to be a C99 project, but if you try to use the header in a C++ source file, GCC 6.3.1 g++ spits out the following.

> 
> In file included from /usr/include/appcontype.h:9:0,
>                  from test.cpp:2:
> /usr/include/cipconnectionmanager.h:153:26: error: duplicate ‘const’
>    const ConnectionObject const *connection_object);
>                           ^~~~~
> /usr/include/cipconnectionmanager.h:158:57: error: duplicate ‘const’
>  CipUint GetProductionInhibitTime(const ConnectionObject const *connection_object);
>                                                          ^~~~~
> /usr/include/cipconnectionmanager.h:164:26: error: duplicate ‘const’
>    const ConnectionObject const *connection_object);

Apparently C++ and C handles the const qualifiers differently with incomplete types.

The simple thing to fix this and the only thing I know of on how to fix this is to remove one of the const qualifiers but I'm not sure if that is an acceptable solution.

simonlui, 2017-01-12:
Also, this is slightly unrelated but the function prototype in cipcommon.h for EipStatus NotifyClass(const CipClass *RESTRICT const class, ...) doesn't match the function signature in cipcommon.c which is EipStatus NotifyClass(const CipClass *RESTRICT const cip_class, ...). The header also causes problems also as class is a reserved word in C++.

CapXilinx, 2017-01-12:
To the duplicate const issue, could you try to modify
`const ConnectionObject const *connection_object`
to
`const ConnectionObject  * const connection_object`

and check again if the error is showing up?

To you second point, thank you for the hint, I will change that.

simonlui, 2017-01-12:
Sorry for the late reply, but yes, the error disappears with your modification.

CapXilinx, 2017-01-12:
Thanks for checking!

Yes, this was my guess, as I used this type of const pointer to const object in C++ a lot. Unfortunately GCC 4.9.4 does not implement a warning for this, which is why this slipped my attention.

commit 2d4f3bba (2017-01-12): Merge pull request #81 from CapXilinx/master

Fixes #80 fixes const position in function arguments, renames parameter class to cip_class
commit 7faccb4f (2017-01-12): Fixes #80 by correcting position of const qualifier
closed 2017-01-12 by CapXilinx