EIPStackGroup/OpENer #82: CompactLogix L43 running version 17.12 with a EtherNet/IP module 1768-ENBT fails to establish a connection

simonlui, 2017-01-13:
Sorry for putting in so many issues in such a short time. Here's the trace log for this.

> creating class 'message router' with id: 0x2
> adding 1 instances to class message router
> creating class 'identity' with id: 0x1
> adding 1 instances to class identity
> creating class 'TCP/IP interface' with id: 0xF5
> adding 1 instances to class TCP/IP interface
> creating class 'Ethernet Link' with id: 0xF6
> adding 1 instances to class Ethernet Link
> creating class 'connection manager' with id: 0x6
> adding 1 instances to class connection manager
> creating class 'assembly' with id: 0x4
> adding 1 instances to class assembly
> adding 1 instances to class assembly
> adding 1 instances to class assembly
> adding 1 instances to class assembly
> adding 1 instances to class assembly
> adding 1 instances to class assembly
> networkhandler: new TCP connection
> networkhandler: opened new TCP connection on fd 6
> Data received on tcp:
> reply sent:
> Data received on tcp:
> reply sent:
> Data received on tcp:
> NotifyMessageRouter: routing unconnected message
> NotifyMessageRouter: calling notify function of class 'connection manager'
> notify: found instance 1
> notify: calling ForwardOpen service
> ForwardOpen: ConConnID 0, ProdConnID 0, ConnSerNo 8
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> We have a Non-Null request
> We have a Non-Matching request
> key: ven ID 0, dev type 0, prod code 0, major 0, minor 0
> classid 4 (assembly)
> Configuration instance id 151
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> assembly: type bidirectional
> connection point 150
> connection point 100
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> networkhandler: UDP socket 7
> networkhandler: bind UDP socket 7relevant or not due to older CIP standard 
> No T->O Sockaddr info available
> no suitable addr info item available / O->T: 0, T->O: -1, Selector: -1, direction: 1
> error in OpenMulticast Connection
> connection manager: connect failed
> AssembleForwardOpenResponse: sending error response
> notifyMR: notify function of class 'connection manager' returned a reply
> reply sent:
> Data received on tcp:
> NotifyMessageRouter: routing unconnected message
> NotifyMessageRouter: calling notify function of class 'connection manager'
> notify: found instance 1
> notify: calling ForwardOpen service
> ForwardOpen: ConConnID 0, ProdConnID 0, ConnSerNo 8
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> We have a Non-Null request
> We have a Non-Matching request
> key: ven ID 0, dev type 0, prod code 0, major 0, minor 0
> classid 4 (assembly)
> Configuration instance id 151
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> assembly: type bidirectionalrelevant or not due to older CIP standard 
> connection point 150
> connection point 100
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> Connection type: 4000 / network connection parameter: 4826
> Connection type: 2000 / network connection parameter: 2822
> networkhandler: UDP socket 8
> networkhandler: bind UDP socket 8
> No T->O Sockaddr info available
> no suitable addr info item available / O->T: 0, T->O: -1, Selector: -1, direction: 1
> error in OpenMulticast Connection
> connection manager: connect failed
> AssembleForwardOpenResponse: sending error response
> notifyMR: notify function of class 'connection manager' returned a reply
> reply sent:
> ...

I did a git bisect and the commit https://github.com/EIPStackGroup/OpENer/commit/70928ce2e52e49f8004e56782fc76efb659dfe81 broke this. I looked at previous issues and this seemed to be part of #48.

I'm not exactly sure if the CIP specification was changed to be not backwards compatible for a situation like this. And unfortunately, I am restricted to this particular firmware version. So I am wondering if this issue is valid or not for this to be fixed.

CapXilinx, 2017-01-13:
Hmm, my first guess would be that I am handling something wrong :)
I don't think that the Forward Open is not backward compatible. Could you provide me a Wireshark trace of the Forward Open request so that I can see the parameters sent by your PLC?

simonlui, 2017-01-13:
I have attached the log.

[log.zip](https://github.com/EIPStackGroup/OpENer/files/704901/log.zip)

CapXilinx, 2017-01-13:
Short answer: Yes, I plan to make a new stable release in the near future, after some code relics are removed from the code base.

Longer answer: I usually tried to provide a working version in the master branch, but since the last official version I got a new version of the EIP specification, so I needed some time to bump it from the 2009 spec to the April 2015 one. There are some last cleanups to make in the code, then I will make an official release and will probably continue with a git-flow like release and development model.
The current version though should be quite match the specification quite well, beside this bug here, as my EIP test tool runs it with only one error (for the type 1 reset). The only other thing I need to check is a warning, that the tool needs a Point-to-point consumer connection for a test.

rubensolozabal, 2017-02-06:
Hi simonlui, 

I had the same problem  --> "No T->O Sockaddr info available"

I think it is solved in **Pull request #86**: Bug open multicast connection 

Just one parameter is not specified before calling OpenMulticastConnection()

simonlui, 2017-02-06:
Hello rubensolozabal.

I checked out your bugfix branch and it solved this issue for me. I suppose this is a mandatory parameter hence the error when trying to make a connection.

I haven't had time to really dig deeper into the code to find the issue since I had to work on some other things at work. Thanks for finding the time to track this down and for creating the pull request.

CapXilinx, 2017-02-12:
Hi!

As the patch from rubensolozabal fixes your problem and is now merged into the master branch, I close this Issue

Also thanks from me rubensolozabal!

Cheers

closed 2017-02-12 by CapXilinx