EIPStackGroup/OpENer #93: How address assembly objects in sample app?

beville, 2017-09-08:
Hi, 

I'm a real novice with Ethernet/IP, so this is a probably a really simple problem to solve.

I built the library on Ubuntu workstation, and have the "OpENer" test app running.  I'm using another open source project, the python-based "cpppo" (https://github.com/pjkundert/cpppo), to run a command-line tool to get the identity of the OpENer test app, so it all seems to be working at a basic level.

My problem now is that I'm not sure how to address the assembly objects created by the test app.  Using cpppo, I can address attributes by tag strings ("foo"), or by class/instance/attribute (e.g. "1/4/5") .  The only specific values I see in the code are the values for the assembly instance numbers:

```
#define DEMO_APP_INPUT_ASSEMBLY_NUM                100 //0x064
#define DEMO_APP_OUTPUT_ASSEMBLY_NUM               150 //0x096
#define DEMO_APP_CONFIG_ASSEMBLY_NUM               151 //0x097
#define DEMO_APP_HEARTBEAT_INPUT_ONLY_ASSEMBLY_NUM  152 //0x098
#define DEMO_APP_HEARTBEAT_LISTEN_ONLY_ASSEMBLY_NUM 153 //0x099
#define DEMO_APP_EXPLICT_ASSEMBLY_NUM              154 //0x09A
```
Are there implicit class and attribute numbers also associated with these assembly instances, so that I can use the class/instance/attribute scheme?  Or am I completely off-base, and maybe cpppo can't talk to the OpENer test app?

Thanks!

azoitl, 2017-09-08:
hi,
no you aren't completely off base here. the CIP spec defines for all CIP objects a set of attributes that need to be supported and a list of optional. For an assembly object instance at least to attributes need to be supported. This is attribute 3 which is a byte array and attribute 4 which is the size of the byte array. 

I hope this helps.

beville, 2017-09-08:
Thanks, that's a huge help, and that pointed me in the right direction.  I didn't realize that assembly objects were special.  The terms "Ethernet/IP" and "CIP" make for tough web searching... :-)

I see now that  all assembly objects have the same class of 4.

I still haven't figured out how to query using the python library, but I was able to talk to the OpENer test app using a windows-based tool, and to set and get attributes at 4/100/3 and 4/100/4.

Thanks, again!

azoitl, 2017-09-09:
Great that this helped you. If you are new to EtherNet/IP and CIP I can recommend the getting start documentations (e.g., the CIP overview) from the ODVA webpage.

closed 2017-10-02 by CapXilinx

remusmp, 2018-12-11:
You can get an attribute in python like below:
```
python3 -m cpppo.server.enip.get_attribute -a 10.0.2.128 -S "@4/100/3"
```

"@4/100/3" means: assembly objects are identified by class id 4, 100 is the instance id (defined in the sample app), attribute 3 is the one containing the value. If you ask for attribute 4 "@4/100/4" then you'll get the size of the value.