EIPStackGroup/OpENer #94: Porting to standalone using lwip

prsullivan, 2017-09-13:
Is there way to port OpENer to a standalone or bare-metal application using lwip as the TCP/IP stack without too much difficulty? It looks like at a minimum the generic network handler would need a rewrite because it relies on a BSD socket api. Would any other code need to be adapted to work with lwip's callback style api?

( https://savannah.nongnu.org/projects/lwip/ )

mmprestine, 2017-09-13:
I have used LWIP with great success on STM32 micros with GCC.  There is
little effort needed to get it working.

CapXilinx, 2017-09-14:
If you would provide me the needed code changes, I could integrate them into OpENer, or at least update the networkhandler to better support lwip.

prsullivan, 2017-09-14:
If lwip is being used on an embedded system with an RTOS, then there is already a compatible socket api and I agree with mmrprestine's sentiment that it would take little effort. However, on the embedded system I am working with, there is no OS and lwip is running in what the lwip documentation calls "Mainloop Mode" and only the "raw" or "Callback" style api of lwip can be used. Lwip's raw api is based entirely event based. A more in depth description can be read here: http://www.nongnu.org/lwip/2_0_x/raw_api.html 

I think the only function that might need some modification is NetworkHandlerProcessOnce().

CapXilinx, 2018-11-09:
As nothing was added to this issue since last year, I now close it.

closed 2018-11-09 by CapXilinx