EIPStackGroup/OpENer #95: g_encapsulation_inactivity_timeout: seconds or milliseconds?

scottnewell, 2017-09-16:
Line 63 of file ciptcpipinterface.c: the comment specifies the timeout as 120 s, but the code considers it as 120 ms. I suspect 120 s is correct, but I'm unsure. Thoughts?

blacktigersoftware, 2017-09-16:
scottnewell -- could you email me at kdavis at rsi-xray dot com?  I would like to see how  you are using this.  I'm in the industrial area, and we were looking at Ethernet/IP for point-to-point communications between various hardware automation boards that we build.  No decision has been made but this is far better than spending thousands on the odva spec. lol.

CapXilinx, 2017-09-18:
Sorry, I was on a business trip, I will look into this issue and your pull request today.

Thanks for your help!

scottnewell, 2017-09-22:
blacktigersoftware--email sent! (Twice.)