EIPStackGroup/OpENer #113: is it necessary to use cmake >= 3.7.2?

luntik2012, 2017-10-03:
I cannot build OpENer in Ubuntu 16.04 after cloning because you require cmake version >= 3.7.2 (while Ubuntu 16.04 has 3.5.1 for now).
But changing to `cmake_minimum_required( VERSION 3.5.1 )` helps, I can build it.

CapXilinx, 2017-10-03:
If 3.5.1 is working, than it is ok for me. I just could not find out, which version is needed for IN_LIST

CapXilinx, 2017-10-03:
And Gentoo already supports 3.7.2 ;)

simonlui, 2017-10-04:
Hi,

Seems like IN_LIST was introduced in 3.3, but unfortunately, I don't have a machine with that old of a cmake version.

https://cmake.org/cmake/help/v3.3/policy/CMP0057.html

CapXilinx, 2017-10-05:
Hi,
thanks! I will check and update the cmake file accordingly

CapXilinx, 2017-10-05:
Fixed in commit 48111f37ebf309897061424fec35acb9e7d03738

closed 2017-10-05 by CapXilinx