EIPStackGroup/OpENer #125: CT15 Conformance Test, Failure to parse message path "Member" type segment

wrc-jasonw, 2017-12-27:
Hello, I work for a company that evaluated using OpENer for developing a product line. Since it didn't pass the ODVA (CT14/CT15) conformance test out of the box we ended up going with a different (much more expensive and seemingly inferior) solution. However, since I did the work I would like to share some of my findings.

Using version CT15 of the conformance test software I encountered an issue. During one step  the conformance software tests which services are implemented for each CIP object. During this test a request with the get member service is made. OpENer fails to parse the request path because it contains an additional segment.

The failing request path contains four elements: class, instance, attribute, and member rather than the usual three: class, instance, attribute.

If the path contains a "Member" type segment it is parsed incorrectly and an error is thrown "wrong path requested" followed by "NotifyMessageRouter: error from createMRRequeststructure."

As a result DecodePaddedEPath() incorrectly parses the path -> CreateMessageRouterRequestStructure() returns an error code -> NotifyMessageRouter() fails, prints the error message, and the conformance test fails...

A wire-shark capture is available, see frame numbers 14 and 15: [opener_bug.pcapng.zip](https://github.com/EIPStackGroup/OpENer/files/1589105/opener_bug.pcapng.zip)

CapXilinx, 2017-12-27:
Hi wrc-jasonw,
thank you for sharing your findings. Such feedback is invaluable for me and the development of OpENer.
Unfortunately I currently only access to the CT14 conformance tool, and I am currently working on eliminating the two errors left (although they appear several times).

Regarding the Member segment, this is probably because OpENer does not yet support Get Member or Set Member services, which is why the parser is not able to interpret the member section. I will try to fix this as soon as the other two errors from CT14 are fixed.

Perhaps after this changes are implemented your company can again check on the OpENer, as otherwise it seems to work very well, even passing the non-device related tests of the ODVA PlugFest.

CapXilinx, 2018-04-24:
Related to Issues #138 #139 #140 #143

commit 769281ed (2018-04-24): Fixes #125 #138 #139 #140 #143

A hack fix, which enables the parsing of Member sections, but due to the
fact, that no serivce exists that uses this data, the data is discarded.

In order to really fix this issue the way how the NotfiyMessageRouter
function works has to be changed from scratch
closed 2018-04-24 by CapXilinx