EIPStackGroup/OpENer #136: Need an IO test

alexhjf, 2018-03-05:
Hi,
I have run OpENer on linux, and I want to test the IO read/write function, however I don't have an IO scanner program
So I wonder if I could use the OpENer API to implement an IO scanner, or what's better, could you offer me a demo to test the IO function of OpENer? It'll be very thankful if you could offer some help :)

CapXilinx, 2018-03-06:
Hi,
as you have surely seen in the other issues related to scanner functionality, OpENer has no scanner functionality.
You can probably part of OpENer as a base for a scanner device, but as OpENer is intended as an adapter device, most of the behavior will not suit for a scanner. As other people have pointed out in several issues, you could try the EtherNet/IP explorer, which can be found at sourceforge.

closed 2018-03-07 by CapXilinx

alexhjf, 2018-03-08:
@CapXilinx
 I have tried the EtherNet/IP explorer but it didn't help testing the I/O functionality of OpENer.
Other scanner simulation software like EIPScan from IXXAT is so expensive that I can't afford it
I'm a novice in the field of Communications and I am not able to get a PLC device either :p 
Do you know some other ways to get a scanner simulation software like EIPScan?
Thank You!

CapXilinx, 2018-03-08:
Unfortunately no, I don't know of any free EIP scanner software

mmasenlle, 2018-12-06:
Hi,

Here 
https://github.com/mmasenlle/samples/tree/master/enip 
there are some very basic "scanner" scripts in python that I use to test adapters.