EIPStackGroup/OpENer #144: Invalid Segment in Connection Path

fchaxel, 2018-04-08:
Hi,

Why OpenEr reject Forward open (Invalid Segment in Connection Path) with the following request. Using path compression gives the same behaviour. Wireshark capture attached.

Same also if only the Instance 100 is request with a O->T set to null and for only Instance 150 with T->O set to null.

As I understand in the CIP specification path could contains 1, 2 or 3 path 

- 1 path O->T or T->O if the other connection type is null and no configuration data provided
- 2 path O-T and T->O if no configuration data provided
- 2 path Config and T->O or O->T if configuration data are provided
- 3 path Config and T->O and O->T if configuration data are provided

Bye

[Eip.zip](https://github.com/EIPStackGroup/OpENer/files/1887079/Eip.zip)

CapXilinx, 2018-04-09:
Hi,

this is because currently can only cope with full compressed EPaths. If path compression also results in the same behavior, please send me a Wireshark trace from that scenario also.

Fully compressed EPaths should work, as the Conformance Test tool also uses these, and checks all scenarios as far as I know.

Best regards,
Martin

fchaxel, 2018-04-10:
Hi Martin,

OK.

As I can see OpenER can handle Forward Open with a 3 instances path only : Config + Output + Input .. that's maybe why I can see dummy Instance 152 & 153.

It will be fine if it could accept ForwardOpen like this (as explain in Table 3-5.13 Encoded Application Path Ordering)

- 20 04 2C 64 only path when 'O->T Connection Type' parameter is set to null
- 20 04 2C 96 only path when 'T->O Connection Type' parameter is set to null
- 20 04 2C 96 2C 64 without config reference (no data are sent for config, so config assembly is not required at all).

24 in place of 2C also (but already accepted by OpenER as request in §5-5.5.3).

Best regards,
Frederic

CapXilinx, 2018-12-06:
Hi Frederic,

I looked into your issue, and I am not completely sure what your use case is. Could you please explain in more detail, what you want to achieve?
Also Instance 152 and 153 are not dummy instances, but the needed heartbeat-assemblies for input only and listen only connections.

Best regards,
Martin

CapXilinx, 2019-07-01:
Closed due to inactivity.

closed 2019-07-01 by CapXilinx