EIPStackGroup/OpENer #145: Max_Instance & Number_Of_Instances static class attributes

fchaxel, 2018-04-10:
Hi,

The two parameters are not OK in OpenER sample application

**Identity class**
Max_Instance = 1
Number_Of_Instances = 7 (should be 1 certainly)

**MessageRouter**
Max_Instance = 0 (should be 1 or more)
Number_Of_Instances = 0 (should be 1 or more)

**Assembly**
Max_Instance = 6 (should be 154)
Number_Of_Instances = 6 

**ConnexionManager**
Max_Instance = 1
Number_Of_Instances = 7 (should be 1)

Bye

CapXilinx, 2018-04-10:
Hi,

I remotely remember there was a special case for the assembly object, that the instance ID is not the real instance ID but the connection epath. But I will check again.

For the other object, how do you get these numbers? In the source, as well as in the eds file the correct values are entered.

Regards
Martin

fchaxel, 2018-04-10:
Hi,

Values are coming from both Ethernet/IP Explorer (sourceforge) confirmed by Molex EIP_Tool (I don't remember from where I get it free).

Bye

CapXilinx, 2018-04-10:
Hi,

I cheched it right away, it seems that there is a bug in the GetAttributeAll service, which returs 7 for attribute 3. If you perform a GetAttributeSingle, the correct value (1) is returned.

Thanks for reporting!
Best regards,
Martin

CapXilinx, 2018-07-12:
Hi, I finally had some time to dig into this issue.
OpENer is performing fine as it is.
For the Identity Object and Connection Manager Object: The GetAttributeAll answer was changes, attribute 3 is not sent anymore, so this is a problem of the Molex ENIP tool. The reported 7 is the Max ID of class attributes, which is correct.

Our message router does not support a GetAttributeAll, so we are sending the default answer, which again does not send data on attribute 3, so this is correct too.

Only the assembly object instance number is incorrect, as you pointed out.

fchaxel, 2018-07-13:
Hi,

I don't have access to spec & tools during some weeks, but if an attribute is not present in GetAttributeAll response the default value must be sent since the receiver cannot identifiy missing element (or attributes after the missing one should not be sent).

Bye.

CapXilinx, 2018-07-13:
Hi,
that's correct, but if you take a look at the specification you will see, that the GetAttributeAll answers of Identity, Connection Manager, and Message Router are specified, that only certain attributes shall be reported and not all available. As said before Attr 3 is not to be sent anymore according to the specification, but Attributes 1,2,7

fchaxel, 2018-08-22:
Hi,

I disagree this view (reading old specification from 2007, maybe changed after)

In chapter
   4-4.1 Class Attributes
       we can read
          Important: If a Class Attribute is optional, then you must define a default value
In chapter
   4-5.1 Get_Attributes_All Response
       we can read
          Important: Insert default values for all unsupported attributes.
and in all chapters concerning classes specification such as
   5-2 Identity Object
       we can read someting like this
          3 thru 7 These class attributes are optional and are described in Chapter 4 of this specification.

So to send attribute 7, all attributes 1 to 6 should be in the response to GetAttributeAll (with a good or a default value).

Bye

closed 2018-08-22 by fchaxel

CapXilinx, 2018-08-22:
Hi!
You are free to disagree, but my in my Vol1 version 3.24, it clearly state a differing behavior for e.g. the Identity objeci in 5A-2.3.2, and as the CT-15 conformace tester is fine with this implementation I am pretty sure, that OpENer is behaving fine.

Bye