EIPStackGroup/OpENer #153: What all things apart from OpENer stack i have to have to start Ethernet/IP running on my BBB with linux

shrikantdhapke, 2018-08-08:
Hi,
I am using Beagle bone black with Linux 4.10.17. I have downloaded OpENer stack and built it with my Linux for beagle bone black. Once i get the command prompt after the board boots,  I am running the OpENer binary as "OpENer eth0". Is this the correct approach or i need to make some other configurations/initialization on my own?

Thanks.

CapXilinx, 2018-08-10:
Yes, thats the correct approach to run OpENer. Configurations for the IP interface have been removed in favor for automatically reading the needed information via the Linux and/or Windows API.
With this you start a adapter device, with an input, output, config, explicit and heartbeat assemblies for listen and input only connections.

CapXilinx, 2018-08-11:
But you can configure device name, vendor ID, device ID, and device product name via CMake settings, if you need different settings as the standard ones.

closed 2018-08-14 by CapXilinx