EIPStackGroup/OpENer #154: TimeOut

MatthewKinsman, 2018-08-29:
I keep getting connection timeouts.  My PLC connects to the PC and the connection enters run but shortly after it drops.  I am attempting to run at 50ms RPI but I have cranked it all the way up to 5000ms and it still drops.  It appears that there may be a issue with the timeout calculation.  I ran wireshark alongside and the packets were received at the correct time.

azoitl, 2018-08-29:
I had similar issues in the past. Most of the time the reason was not correctly used port or multicast addresses. What kind of connection are you using? Do you see packets in both directions? Who is timeouting the PLC or OpENer?

MatthewKinsman, 2018-08-29:
I am using unicast, but multicast appeared to respond the same.  I do see packets in both directions.  It looks like OpENer is the first to time out.  I pulled the deploy branch and it seems to be working better.  I was never able to get the Master branch to establish connection.  The PLC would start the connection but before it went into Running OpENer would time out. I saw this happen on 3 machines (Windows 7, Ubuntu Guest VM, Debian 9) with the same result.  I pulled the develop branch and I was able to get Windows 7 machine to connect fine, The Ubuntu VM would connect but shortly after timeout(I think this actually has to do with the VM though) and Im off to test the Debian machine.

CapXilinx, 2018-08-29:
Okay, that's strange as I have tested the master with the conformance tester 14 and even attended the December PlugFest in Germany. The develop branch has several fixes for the current version of the specification but is not really operational currently. What PLC are you using?

CapXilinx, 2018-08-29:
Could you perhaps also try the tagged version. I just have seen that I added a quick fix after the PlugFest for some in principle unrelated issue

CapXilinx, 2018-08-29:
If this is possible for you, you can also post a Wireshark trace. Perhaps there is some error in the communication.

azoitl, 2018-08-29:
When starting with OpENer I often got situations like the ones you are describing. In the beginning I also thought it is a timing issue of my machine. But it was really always the case that there was a configuration issue. I would expect that the messages coming from the PLC are not in the way as either defined in the forward open or in the forward open response. So that OpENer can not update the connection timers correctly as it is not relating the received messages to the connection.  As Martin pointed out a trace containing the forward open the forward open response and the messages in both directions could help to shade some light on this issue.

MatthewKinsman, 2018-08-29:
I am using a Logix 1756-L71S through a 1756-EN2T.  I have also been using a L16 and got the same issues.  I tried the Deploy branch on my Debian machine but it still seems to be acting up.  I also re ran the Windows Master and attached the Wireshark Capture and Output of the Trace.  This configuration does not ever show connection on the PLC
[OpENer-trace.zip](https://github.com/EIPStackGroup/OpENer/files/2333091/OpENer-trace.zip)

MatthewKinsman, 2018-08-29:
Here is the trace from the Debian Machine.  When I connect I can clearly see the send happen at 1s intervals(the RPI i setup) but it times out.  I never see a sign of receive
[log.txt](https://github.com/EIPStackGroup/OpENer/files/2333158/log.txt)

MatthewKinsman, 2018-08-29:
Turns out the second bind in CreateUDPSocket causes the socket to not ever Set in the select() on my Linux machines. I commented out and my Debian machine is now working

CapXilinx, 2018-08-30:
Hi!
Thanks for the fix. Yes there is a second erroneous bind inside the code, which I realized is not doing what I intended it to do, after commit it. The current issue I want to fix, and the reason for this bind, is that the current conformance tester checks if packets are sent from the configured or default source port , which is 2222. So I need to share certain sockets and didn't realize that even when I am not reading from one socket, the other socket will not get all messages sent to the port.

CapXilinx, 2018-08-31:
Ok, so as it seems to work for you now, I close this issue.
Thanks again for the help and fix!

closed 2018-08-31 by CapXilinx