EIPStackGroup/OpENer #156: Facing ENOTCONN error while usig the OpENer stack with linux kernel 4.14.51

shrikantdhapke, 2018-09-04:
Hi,
I am evaluating the OpENer stack on Beagle bone black(a8). After my board boots i run the OpENer binary as "./OpENer eth0". Following the sequence of function calls once it tries to get the network interface details from the eth0 name, i am getting error in CloseSocket(fd) function call which gets called from function "ConfigureNetworkInterface()" in file "ports/POSIX/networkconig.c" in . It gives me the below error:
Could not close socket 4 - Error Code: 107 - Transport endpoint is not connected

Is there any known issue with kernel 4.14.51 OR anything that you can suggest can cause this issue?

Regards,
Shrikant

CapXilinx, 2018-09-04:
I have the same error. I think it is a bug, but this should have no consequences for the further functionality of OpENer. Most likely this comes from the shutdown command of close socket, as the socket needed to extract the network interface parameters is not connected, and therefore cannot be shut down.

shrikantdhapke, 2018-09-04:
Ok, Thanks for the reply. Is it going to be fixed in next version of stack?

CapXilinx, 2018-09-04:
sure, but it is currently not very high on my priority list, as I first have to fix an issue with socket handling for the new tested source ports

shrikantdhapke, 2018-09-05:
ok. Thanks.

closed 2018-09-05 by shrikantdhapke