EIPStackGroup/OpENer #160: Fails after broken connection and reconnect

jkvatne, 2018-10-24:
When the connection between the scanner and my PC is broken (Network cable unplugged) the OpENer times out as it should. When I reconnect the cable, the scanner sends a new Forward Open request and the OpENer returns success. But it does not close the old connection.

I have traced this problem to a missing ! in appcontype.c:203. (the call to `EqualConnectionTriad() `). I think we should close the connection if it is NOT equal.

I do not completely understand the logic here, but my driver now reconnects fine after cable plug out and in.

CapXilinx, 2018-10-24:
Thanks for tracing the problem down. I know the logic is quite complicated, I will give the proposed fix a try, and will test it with the conformance tester.

Best regards,
Martin

CapXilinx, 2018-10-31:
Hi,
thanks again for digging into the problem, but I think your fix is wrong, as only the original owner should be able to claim the connection, and by negating the EqualConnectionTriad you will only specify that it has to be not the original connection, which will not occur anyhow due to the timeout.
I think there are two bugs that play in concert, first is should not be EqualConnectionTriad, but ConnectionObjectEqualOriginator, and second if a timed out exclusive owner is still registered and a different originator wants to establish a connection the new connection would have to be declined as an ownership conflict. As this check or returning the corresponding error code was missing, your driver was able to reconnect with a new connection, but could not reclaim the original one. I already have a fix and as soon as my CT tool is happy with the new version I will push it on the develop branch.

Could you then test it if this solves your problem and let me know?

Best regards,
Martin

jkvatne, 2018-10-31:
Hi, and thank you for the help

The fix does in fact work fine for me.  There is only a very small issue left, when I recoonect, the error trace prints "Attempt to provide a NULL pointer to node pointer". This seems not to mage any problems, and I have not attempted to find out why the message is printed, as my application works fine - also after many reconnects.

Best Regards
Jan Kåre

commit 3bbb2a14 (2018-10-31): Fixes #160 - Changes check in appcontypes to same originator

Previous in the appcontypes, while getting a connection, it was checked
if a timed out connection had the same connection triad. This was wrong
as a timed out connection is not existing anymore, and each connection
has a unique connection serial number. In order to enable that the
original originator and only the original originator can reclaim the
timed out connection, first the check has been changes to equal
originator (vendor ID and serial number) and in the case that it is not
the original originator an ownership conflict error [... cut]
closed 2019-01-14 by CapXilinx