EIPStackGroup/OpENer #163: Clarification of what is permitted under the license

shiftee, 2018-10-26:
Do I need to obtain a license from ODVA to use this library if I do not use the EtherNet/IP mark?

From the extract below it seems I don't but it would be nice to clarify:

`
The Communications Stack Software for EtherNet/IP, or any portion thereof, with
or without modifications, may be incorporated into products for sale.  However,
the software does not, by itself, convey any right to make, have made, use, 
import, offer to sell, sell, lease, market, or otherwise distribute or dispose 
of any products that implement this software, which products might be covered 
by valid patents or copyrights of ODVA, Inc., its members or other licensors 
nor does this software result in any license to use the EtherNet/IP mark owned 
by ODVA. To make, have made, use, import, offer to sell, sell, lease, market, 
or otherwise distribute or dispose of any products that implement this software, 
and to use the EtherNet/IP mark, one must obtain the necessary license from 
ODVA through its Terms of Usage Agreement for the EtherNet/IP technology, 
available through the ODVA web site at www.odva.org. This license requirement 
applies equally (a) to devices that completely implement ODVA's Final 
Specification for EtherNet/IP (“Network Devices”), (b) to components of such 
Network Devices to the extent they implement portions of the Final 
Specification for EtherNet/IP, and (c) to enabling technology products, such as
any other EtherNet/IP or other network protocol stack designed for use in 
Network Devices to the extent they implement portions of the Final 
Specification for EtherNet/IP. Persons or entities who are not already licensed
for the EtherNet/IP technology must contact ODVA for a Terms of Usage Agreement.
`

Thanks,
Mark

CapXilinx, 2018-10-29:
From my understanding, as soon as you distribute software or products of any kind based on OpENer you have to obtain a license from ODVA, regardless if you use the EtherNet/IP mark or not. The open source license of OpENer was created in cooperation with the ODVA, so if you need a clear answer to this, I can try to contact the ODVA to get their opinion on the exact terms of the license.

shiftee, 2018-10-30:
Thanks, I think it would be useful to have a clear answer to this question.

These articles [article1](https://www.theengineer.co.uk/supplier-network/product/rockwell-sponsors-open-source-ethernetip-stack),  [article2](https://www.automation.com/library/resources/free-open-source-ethernetip-communication-stack) mention:

> developers can download the new license and royalty-free adapter stack through GitHub.com

If a separate license must be obtained then these statements are misleading, and I believe the software doesn't qualify under the Open Source trademark

azoitl, 2018-10-31:
I have to disagree here with you. The stack itself is royalty free. However the technology the stack is implementing, i.e., CIP and EtherNet/IP are not. This is true for different technologies out there (e.g., IEEE 1588). Still you get quite some advantages of OpENer as you can experiment and test with it, have the source available open for adjustment to your needs, and the shared development risk.

shiftee, 2018-10-31:
So you can access the source code but are not free to use it in a product.

One needs a license to use the code commercially because the protocols are patented?

Are the previous two statements accurate?

azoitl, 2018-10-31:
As Martin said we both are no layers. But as understand the situation it is as you said.

Generally speaking it is very hard to openly implement all these "open" industrial protocols.

shiftee, 2018-11-05:
So it seems we have clarified the depressing truth of the matter.

Have the ODVA been contacted for an official response?

If so then we can wait for a response, otherwise we should perhaps close this issue?

Thank you both for your help,
Mark

closed 2018-11-09 by CapXilinx