EIPStackGroup/OpENer #164: A basic question about OpENer

alexhjf, 2018-10-27:
Hi,
I'm a new developer of EIP, I want to know that if OpENer could be implemented to be a explicit message server and client ? Or it's just an I/O adapter ?
Thank you!

azoitl, 2018-10-28:
as an I/O adapter requires an Explicit message server OpENer is automatically also an explicit message server and you can set it up quite easily. Theoretically much code in OpENer can also be utilized for explicit message clients. But for these much more work has to be done (e.g., code for connecting to severs, assembling requests, processing responses).

alexhjf, 2018-10-31:
Hi, azoitl
Yeah I've learned that from reading "_A Guide for Ethernet/IP Developers_" of _ODVA_ .
So just like a typical EIP adapter, OpENer has the functionality of Unconnected message Server/Client, Connected message Server and I/O Server. Am I right ?

CapXilinx, 2018-10-31:
Hi,
no a typical EIP adapter is an implicit and explicit connection server without any client capabilites. Connected or unconnected messages are only two types of explicit connections. Unconnected message do not establish a CIP connection before sendings commands.

OpENer could still serve as a base for a CIP client, as most of the message structures and the functionality for encoding messages are available for further use.

closed 2018-11-09 by CapXilinx