EIPStackGroup/OpENer #177: Question： Can I data-link 2 or more equipments which using OpENer?

frankgp, 2019-01-29:
Now I want using ENIP to connect some equipments through data-link. 
Can I use the OpENer to connect 2 or more equipments? 

Sorry that I am a beginer of ENIP.

azoitl, 2019-01-29:
As posted already OpENer is currently just an adapter device. As such it is from the connection establishment point passive. That means other devices can connect to OpENer but not the other way round. However OpENer can be configured such that several devices can connect to OpENer. So if this is your use case than OpENer is your tool.

frankgp, 2019-01-30:
Thank you for your explain. 
Now I understand the OpENer function.   
OpENer stack cannot establish connection with other slaves active. 
So 2 adapter devices with OpENer cannot data-link each other. 
I think "adapter device" is not right. We should call "slave device". Is it right?

azoitl, 2019-01-30:
In EIP terminology classical slave devices are called adapter devices. We name OpENer here as adapter implementation as you will find this term also in all the documents that ODVA provides.

frankgp, 2019-01-31:
Thank you for your explain in detail. 
Now I understand.   I make mistake between communicaition adaper and I/O adapter.
Because I use OMRON PLC, in their configuration tools, call their PLC communication adaper.

Thanks again!

closed 2019-02-08 by frankgp