EIPStackGroup/OpENer #179: Build fails on Cygwin

thiagoralves, 2019-02-04:
Running ./setup_posix.sh works. However, after running "make" the compilations ends at 86% with a header missing: ~/OpENer/source/src/ports/POSIX/networkconfig.c:19:10: fatal error: linux/if.h: No such file or directory #include <linux/if.h>

I figured that this file on Cygwin is located at <net/if.h>. After making the change in networkconfig.c, compilation fails again at 97% with another header missing: ~/OpENer/source/src/ports/POSIX/main.c:9:10: fatal error: sys/capability.h: No such file or directory #include <sys/capability.h>

I couldn't find this capability.h header anywhere on the Cygwin include files.

isaoparra, 2019-02-08:
I have a similar problem on ubuntu 16.04.  it was solved after installing _libcap-dev_

`sudo apt-get install libcap-dev
`

thiagoralves, 2019-02-09:
I saw that when I was googling about this issue. The problem is that libcap-dev is not available for Cygwin. At least I couldn't find it anywhere on the package list

closed 2019-02-20 by CapXilinx

CapXilinx, 2019-02-20:
Hi,
I updated the README regarding this issue, but as two alternatives are provided for Windows, and this would need another config switch in CMake I decided against adding a permanent fix to the code.

If this is really an issue for several people out there, we can discuss this again.

Best regards,
Martin

commit f7c68ce4 (2019-02-20): Closes #179 - Adds description in README for Cygwin

Adds description in README that the capability code has to be removed
for Cygwin, or to use either MinGW or MSVC
commit 2ef5447c (2019-02-20): Closes #179 - Adds description in README for Cygwin

Adds description in README that the capability code has to be removed
for Cygwin, or to use either MinGW or MSVC
commit 94dccba9 (2019-02-20): Closes #179 - Adds description in README for Cygwin

Adds description in README that the capability code has to be removed
for Cygwin, or to use either MinGW or MSVC
commit b10283af (2019-02-20): Closes #179 - Adds description in README for Cygwin

Adds description in README that the capability code has to be removed
for Cygwin, or to use either MinGW or MSVC