EIPStackGroup/OpENer #183: Shared Library

wpmccormick, 2019-03-11:
I'd like to build my device application by linking to shared library's. 

_MyDevice/CMakeLists.txt_ 
```
cmake_minimum_required(VERSION 2.8.12)
add_compile_options(-std=c++11 -w)
set(OpENer_PLATFORM "POSIX" CACHE STRING "Set Platform")
set(BUILD_SHARED_LIBS ON CACHE BOOL "Use Shared Libs")
add_subdirectory(OpENer/source)
install(FILES package.xml DESTINATION share/OpENer)
```
It makes the shared lib .so files, but It'd be nice if it could install them (and header files) somewhere. 

Also, I'd like to be able to link against libPOSIXPLATFORM.so, without having sample_application in the way; so do I need to basically recreate your main.c and link against libPLATFORM_GENERIC.so?

It's not clear to me how to make my own OpENer application unless I follow your template very closely.

Thanks,

Bill

wpmccormick, 2019-03-17:
I'm running into some issue getting shared libraries to work; mainly, the issue is with libCIP.so looking for functions from sample application. Seems to be a circular reference. My C coding experience is probably a limitation at this point. I'd appreciate thoughts and insight from the developer.

My main goal is to use OpENer in a C++ ROS project.

CapXilinx, 2019-03-25:
Merged in commit a3bc070083c1e48ee1fc9490467d49ea198258a9

closed 2019-03-25 by CapXilinx