EIPStackGroup/OpENer #188: Connection Manger test with CT-16 in Development(passed) and Conformance test(failed)

vamshikrishn, 2019-03-27:
Hi All,
We found one  issue while executing the object test cases by Conformance and Development test mode options. 

In Development test mode, while running  Connection manager object test, the tool was prompting to select class and Instance services options  as shown in below screen shot(test run successfully without any errors). In conformance test mode, or if we opt for “run continuously” in development mode,  the tool didn’t ask  such type of option during running of connection manager object due to which the tool thrown connection manager errors after completion of the test. 

After completion of both test modes we observed that why error thrown in conformance mode and why not in Development mode. I think during conformance mode or while opting for “running continuously” in development mode, it should ask for option to select instance or class serviced as shown in below screen shot. 

Can anyone let us know what is the difference between Conformance test and Development test ? since in Development mode its asking for user inputs but in Conformance test mode tool not asking for user inputs(Connection manager tests to run)

Could anyone can please help me on this issue to resolve this issue from our code side ?

Thanks,
Vamshi
[image]

CapXilinx, 2019-03-27:
Hi,

first, this is not an issue regarding OpENer. If there are any issues with the ODVA CT tool, please contact the ODVA.
Second, in development mode you can choose which tests are performed to accelerate the development (hence 'development mode'). The conformance test mode tests, as the name suggests, the conformance of your device to the ODVA standards, so all tests are performed (only profile verification and reset tests have can be skipped)

Best regards,
Martin

vamshikrishn, 2019-03-27:
> Hi,
>
> first, this is not an issue regarding OpENer. If there are any issues with the ODVA CT tool, please contact the ODVA.
> Second, in development mode you can choose which tests are performed to [...]

Hi Martin,

Thanks for your quick response. I have discussed  the same with ODVA people but they are saying there is no issue with CT-16 Tool, need to fix these type of issues from code side only. 

Errors observed in log file after running Connection Manager object:
 1> *** Extended status should not have been returned: All extended status values are reserved unless otherwise indicated within the object definition [Vol 1, Appendix B-1].
 2> *** Incorrect error code, Srv Code x4E, Expected x08
 3> *** Extended status should not have been returned: All extended status values are reserved unless otherwise indicated within the object definition [Vol 1, Appendix B-1].
 4> *** Incorrect error code, Srv Code x54, Expected x08
 1> *** ForwardOpen Returned unexpected Status = 0x1, Extended Status = 0x0315; expected Status = 0x0
               [Vol 1 Ed 3.25 Table 3-5.33].

Please let me know if these type of errors from our code side ?

Thanks,
Vamshi

CapXilinx, 2019-03-27:
Hi Vamshi,

I would guess, yes these are errors originating from the stack, but as I do not have CT-16 available I can't test it. Do you run vanilla OpENer or have you already changed the code?

Best regards,
Martin

vamshikrishn, 2019-03-28:
> Hi Vamshi,
> 
> I would guess, yes these are errors originating from the stack, but as I do not have CT-16 available I can't test it. Do you run vanilla OpENer or have you already changed the code?
> 
> Best regards,
> Martin

Hi Martin,
May i know which tools you will use generally for Development or Conformance test? Please share us the links if it is available in web.
we took the open source code from smkohnstamm-OpENer and changed code according to my switch requirement. 

Thanks,
Vamshi

CapXilinx, 2019-03-28:
Hi,

usually I use the CT tool, which is provided to me by the ODVA under certain conditions, and the Molex ENIP tool. I also started using the ENIP tool from Hilscher.

Could you please provide me a link to the smkohnstamm-OpENer version?

Best regards,
Martin

vamshikrishn, 2019-03-28:
Hi Martin,

we took the code from the path: https://github.com/EIPStackGroup/OpENer/tree/075583f61ae53564a4bbe4f71dce5ede6143643c/source/src/cip

Initially we gone through the code from  smkohnstamm-OpENer but we implemented our code from above link. We have implemented RSTP Port, RSTP Bridge, SNMP and Base object and run successfully with CT-16 Tool. 
May i know which version of CT Tool you are using ? we are having CT-15 and CT-16 Versions(you can get the errors with any of the versions). Can you please share us the links to download Molex ENIP tool if it is available and free version.

Thanks,
Vamshi

vamshikrishn, 2019-03-28:
Hi Martin,

I think, I found the root cause for this issue. we are not handling below requests sent by EIP Scanner(CT Tool) in our code.
As per conformance testing behavior while running Connection Manger Object, it was expecting to handle below mentioned requests in cipconnectionmanager.c.
-->HandleNullMatchingForwardOpenRequest()

-->HandleNullNonMatchingForwardOpenRequest()

Thanks,
Vamshi

CapXilinx, 2019-03-28:
Hi,

I am using CT-15 currently, and I do not get this error. Both of these cases are handled in cipconnectionmanager.c, in both the currect version and in the commit you specified.

I do not have the link at hand, just google it you should find it easily.

Best,
Martin

vamshikrishn, 2019-03-28:
Hi Martin,

Can you please share the cipconnectionmanager.c file relevant to  HandleNullMatchingForwardOpenRequest() and HandleNullNonMatchingForwardOpenRequest(). i want to use in my code and see if it is working for me.

Existing code:

EipStatus HandleNullMatchingForwardOpenRequest(
CipConnectionObject *connection_object,
CipInstance *instance,
CipMessageRouterRequest *message_router_request,
CipMessageRouterResponse *message_router_response
 ) {
  OPENER_TRACE_INFO("Right now we cannot handle Null requests\n");
  return AssembleForwardOpenResponse(
    connection_object,
    message_router_response,
    kCipErrorConnectionFailure,
    kConnectionManagerExtendedStatusCodeNullForwardOpenNotSupported);
}

Thanks,
Vamshi

CapXilinx, 2019-03-28:
Hi,

perhaps I was a little unclear. It is handled by sending an error message, which is valid according to the ENIP specification.

If you need this functionality you would need to implement it.

Best,
Martin

vamshikrishn, 2019-03-28:
Hi Martin,
    Even I thought as per EIP specification, it is not mandatory. But, when I spoke with ODVA members, they said that, there should not be any errors.  But when switch sends error message as a response, then, tool throws error as if it was not expected behavior.
Did you run Connection manager object in  Development test mode by clicking Run Continuously ?
we are facing issues in this scenario. 
 
Thanks
Vamshi

vamshikrishn, 2019-03-28:
Hi Martin,
I mean to say click on Run Continuously in Connection Manager object.
[image]

Thanks,
Vamshi

CapXilinx, 2019-03-28:
Hi,

you are implementing a switch, which could have different requirements as adapter devices. OpENer was designed for adapter devices, and for them these requests are for sure not mandatory.
I attended last PlugFest and this was not an issue there, and this is not an issue for the CT-15 tool in development (+run continuous) and conformance mode (+run continuous) as I test it right now.

Best,
Martin

vamshikrishn, 2019-03-29:
Hi Martin,

Thanks for your inputs. 
If that was the case, then it seems, configuration might be the issue. Can you please share the configuration screenshots for connection manager object? Please find the configuration used for my switch with which I am facing errors.

[image]

Thanks,
Vamshi

CapXilinx, 2019-03-29:
Hi,

my CT config is included in the data folder of the repository.
What I see from your config is, that you use a variable size connection with both a min and max size of zero. I am not an expert on variable data sizes, but that seems counter-intuitive.

Best,
Martin

vamshikrishn, 2019-04-01:
Hi Martin,
Thank you for your inputs. we have discussed with ODVA team, and confirmed that new connection is optional and we can skip that. 
we are facing one error on sequence number in Connection manager object shown below. 

  3> *** The DUT didn't return the previous message for the request with the same sequence count. Please see Vol 1 Figure 3-4.5 Server Transport Class 3 Behavior.

 Root cause:
The specification states the following “If the sequence count is the same as the previous message, notify the Link Producer to produce the previous response including the same sequence count.” [Figure 3-4.5 Server Transport Class 3 Behavior]

This means that if the sequence count is the same as the previous message, our message response should be the same as the previous request. 

In this case, the request is for Attribute 6 (Serial Number), but the DUT should respond to the previous message which was a request for Attribute 1 (Vendor ID) because the sequence id is the same as the previous message. 

Please let me know if you faced this error.

Thanks,
Vamshi

CapXilinx, 2019-04-01:
Hi Vamshi,

no, as this is fixed in the current version.

Best,
Martin

vamshikrishn, 2019-04-01:
Hi Martin,
We are unable to find the relevant issue from the current version. 
It will be really great full if you know and please share us the bug number where this issue is resolved. 

Thanks,
Vamshi

CapXilinx, 2019-04-01:
Hi,

its Issue #141

Martin

vamshikrishn, 2019-04-01:
Hi Martin,

Thanks a lot. I will look into this bug. 

Thanks,
Vamshi

CapXilinx, 2019-04-01:
Ok, as the original issue is solved, I close this issue

closed 2019-04-01 by CapXilinx