EIPStackGroup/OpENer #190: Modbus Object

reddiuday, 2019-04-29:
Hi, 

could you share your inputs  for below query?

1. I have device which support CIP Ethernet/IP Protocol. But it is not able to understand or answer when it received the request as a Modbus request. Therefore am planing to implement EIP converter for to transfer the data EIP to Modbus Device. could you please share what are the necessary files required to implement EIP Converter or if  you have any developed, code could you please share it. 

Thanks, 
Udaya Sekhar

wpmccormick, 2019-04-29:
It sounds like you need a device that supports Modbus, or controller/scanner that does EIP. 

There are gateway products (e.g. Anybus, Prosoft, etc.) that will do the protocol conversion, but really you only should consider a gateway when you have no other options. Again, getting the right device that talks the required protocol is usually the best option. Many device vendors package the same product to support different protocols.

In any event, as this project implements an EIP device (not a scanner), there's probably not much here to help you if you're set on making your own gateway. There are open-source EIP scanner projects out there (e.g. ros industrial odva-ethernetip) and Modbus is well known protocol, so you might be able to bolt something together, but if you just want to make something work ... buy it!

Cheers,

-Bill

CapXilinx, 2019-04-30:
In principle as Bill said, but if you want to integrate Modbus Devices into the CIP architecture, I want to point you to the ODVA CIP specification Volume 7A: Integration of Modbus Devices into the CIP Architecture, which you can aquire at www.odva.org

I did not implement this, and I do not plan to implement this in the future.

Best,
Martin

closed 2019-04-30 by CapXilinx

reddiuday, 2019-04-30:
Hi Martin. 

Thanks for the information. 

Thanks, 
Udaya Sekhar