EIPStackGroup/OpENer #193: how does it work?

mchiasson, 2019-05-08:
I'm new to EtherNet/IP and trying to figure out how to turn a raspberry pi into an Ethernet/IP device using OpENer to make it communicate to other Ethernet/IP devices on the network. After studying `sampleapplication.c`, reading http://eipstackgroup.github.io/OpENer/d2/dc9/group__CIP__API.html and trying stuff around with Molex EtherNet/IP tool, I am completely lost. 

[image]

`AfterAssemblyDataReceived` and `BeforeAssemblyDataSend` are just never called. 

So my questions are:
With OpENer, how do I receive explicit messages from other EtherNet/IP devices? How to I send explicit messages to other EtherNet/IP devices?

CapXilinx, 2019-05-09:
AfterAssemblyDataREceived and BeforeAssemblyDataReceived are application specific functions and are not needed for the basic operation of OpENer.

The best way to start is to use the predefined Object tabs from Molex, but beware, sometimes the results will be "wrong" for GetAttributeAll, as the tool has not been updated for a long time. There is a similar tool from Hilscher, I think it is also called EtherNet/IP tool, which is available here https://kb.hilscher.com/pages/viewpage.action?pageId=97444743.

When you start OpENer it will start listening ports on the specific interface, but is otherwise passive.

So other devices have to send commands to OpENer, and it will answer as required by the specification.

Best,
Martin

mchiasson, 2019-05-09:
I only put breakpoints in AfterAssemblyDataREceived and BeforeAssemblyDataReceived just to see if they get hit with whatever action I tried doing from Molex.

I don't see any predefined Object tabs from Molex:
[image]
Maybe i'm not using the right version? either way, I will try the other tool

I also found that you can host your own custom service, class and instances (from issue #18). Is that how I would get something showing up in Molex under the explicit message tab?

So, if I understand correctly, once you have your assembly and connection point setup, the any other devices (e.g. a PLC) can read assembly data and write to assembly data to my OpENer device. Is there a way to read or write assembly data to other devices (e.g. to a PLC) with OpENer?  If I have two RaspberryPi both running OpENer, can I make them transfer data to each other via OpENer?

Thanks!

CapXilinx, 2019-05-09:
The Tabs right from explicit message are the ones I talked about 😄

No, you do not need custom services for the explicit message tab. You only have to provide service, class, instance, and attribute codes which are supported by OpENer. E.g. the getattributeall service, which is available in the Service combo box

CapXilinx, 2019-05-09:
To your second question, no not without a ENIP Scanner or at least an explicit message client

mchiasson, 2019-05-09:
I see... the only thing that works is Service 1 (getattributesall) class 0x01, Instance 1, which gives me a a green "OK" status. I am seeing some data in the response, but i'm not sure what I am looking at. is it showing the data from `g_assembly_data096` (the data container for `DEMO_APP_OUTPUT_ASSEMBLY_NUM`)?

mchiasson, 2019-05-09:
Ok, i'm pretty sure that this is the "identity" data, since i'm seeing `"Opener PC"` in the response payload :-D. So how do I read the data from the output assembly using OpENer's sampleapplication ?

mchiasson, 2019-05-09:
> To your second question, no not without a ENIP Scanner or at least an explicit message client

So it sounds like OpENer might not be the tool that I need. Do you have any suggestion on other 3rdparty lib (preferably C/C++) that can be used for sending requests to other EtherNet/IP devices?

wpmccormick, 2019-05-09:
You could take a look some of the ROS-I drivers I wrote for Tolomatic
devices.

https://github.com/ros-industrial/tolomatic

mchiasson, 2019-05-09:
@wpmccormick **INTERESTING!!!** Thank you for sharing.

CapXilinx, 2019-05-09:
You can read the sample data via a GetAttributeSingle request on the Assembly Object, instance 96d, attribute 3. There the data should be contained.

To this question: "Is there a way to read or write assembly data to other devices (e.g. to a PLC) with OpENer? "
Yes, that's the main functionality of OpENer. Just open a bidirectional implicit connection (Class 1) to the corresponding Assembly object instance and this should work out of the box. 

If you want to actively send requests, or transfer data between two instances without any furhter software, then no OpENer is not the tool you need.
OpENer is a so called ENIP Adapter device, used e.g. for Remote I/Os and other kind of slave devices. 

One tool I know is the EtherNet/IP Explorer which is a Scanner (master) and is written in C#.
If you have a tool at hand, 

What I never tried, and not really thought about is, that if you have means to send the necessary forward open commands to two OpENer instance, you COULD be able to create a bidirectional Class1 connection between these two instances. But I would have to check the specification if this is really possible.

Another option would be to used the base elements of OpENer to add the explicit message client capabilites. Sending explicit messages should be comparably simple.

mchiasson, 2019-05-10:
Using OpENer to create ENIP adapter devices, and using odva_ethernetip to create ENIP scanner devices, I have everything I need :-)

Thank you @CapXilinx and @wpmccormick

closed 2019-05-10 by mchiasson