EIPStackGroup/OpENer #195: How to get two OpENer apps communicating with implicit message I/O

mchiasson, 2019-05-21:
I tried creating two apps based on https://github.com/EIPStackGroup/OpENer/tree/master/source/src/ports/POSIX/sample_application

The only difference are:
1. I named them two different `OPENER_DEVICE_NAME` and and give them different serial numbers.
2. I swapped `DEMO_APP_INPUT_ASSEMBLY_NUM` and `DEMO_APP_OUTPUT_ASSEMBLY_NUM and their corresponding buffer accordingly in one of them.

Both run on the same isolated network under a router that allows broadcast/multicast. both are running on two different linux PC and both have been setup with ```sudo setcap cap_net_raw+ep ./src/ports/POSIX/OpENer```. Both show up in the Molex app when doing a network broadcast in UDP.  However, no matter what changes I make to `g_assembly_data064` and `g_assembly_data096`, on either side, nothing happens. I was under the impression that it was all going to synchronize the data automatically. Am I missing something?  Thanks

wpmccormick, 2019-05-22:
OpENer enables an Ethernet/IP _DEVICE_; it expected that some Ethernet/IP _SCANNER_ (like a PLC) is part of any complete system. In other words, a single _SCANNER_ communicates with one or more _DEVICES_.

mchiasson, 2019-05-22:
@wpmccormick I didn't bother looking into _odva_ethernetip_ after spotting this comment here in your code https://github.com/ros-industrial/tolomatic/blob/master/acsi_eip_driver/include/servo.h#L115 but I guess I will need to give this a shot and see if it implicit data works. If it doesn't, I'm not sure how I'm going to test the adapter that I'm tasked to implement.

I need to figure out what _sending the necessary forward open commands_ mean here, which is obviously what I am missing: 
https://github.com/EIPStackGroup/OpENer/issues/193#issuecomment-491015565
> What I never tried, and not really thought about is, that if you have means to send the necessary forward open commands to two OpENer instance, you COULD be able to create a bidirectional Class1 connection between these two instances. But I would have to check the specification if this is really possible.

CapXilinx, 2019-05-22:
The forward open is a service, triggering the adapter device to open a communication channel. This is sent by the scanner to initiate a Class1 or Class3 communciation channel

wpmccormick, 2019-05-22:
> @wpmccormick I didn't bother looking into _odva_ethernetip_ after spotting this comment here in your code https://github.com/ros-industrial/tolomatic/blob/master/acsi_eip_driver/include/servo.h#L115 [...]

I never had the time/resources to make the the odva_ethernetip implicit comms work for more than one device; and I'm pretty sure you can make it work for 1 device. For more than 1 device, the application needs to manage multiple local UDPsocket's. The example that I had (omron scanner) didn't do this, so I just used the explicit connection. I think it's probably NOT and an issue with the odva_ethernetip side of things; you just need to manage multiple local UDP sockets,

Also, although the intent of odva_ethernetip is to be used in ROS applications, I don't think it is a requirement if you know how to use the cmake build system. And as Ethernet/IP device interfaces are (obviously) customized for the physical device, it is necessary that the interface classes and assemblies are custom as well. So the Tolomatic drivers should only be used as examples; and really, the only parts you'll reference are the parts that do the serialization.

mchiasson, 2019-05-22:
I just want to simulate my PLC until we receive the actual hardware. I will give `odva_ethernetip` a shot. Thanks

wpmccormick, 2019-05-22:
Have you considered running CODESYS on a Raspberry PI?

It's a pretty low cost solution .. probably < $100.

mchiasson, 2019-05-23:
I can't seem to be able to create a UDP connection using `odva_ethernetip` to the OpENer sample app. 

```
Error:   RR Data Command failed with status 8
         at line 290 in /home/mchiasson/odva_ethernetip/Source/src/session.cpp
"Could not start UDP IO: RR Data Command Failed"
```

Here's the log output from `OpENer`:
```
Entering HandleDataOnTcpSocket for socket: 15
Data received on tcp:
Handles data for TCP socket: 15
Send Request/Reply Data
NotifyMessageRouter: routing unconnected message
NotifyMessageRouter: calling notify function of class 'connection manager'
notify: found instance 1
notify: service 0x5b not supported
notifyMR: notify function of class 'connection manager' returned a reply
Sets time stamp for socket 15
TCP reply sent:
Sets time stamp for socket 15
```

is it happening because OpEner doesn't support service 0x5b not supported?

mchiasson, 2019-05-23:
is there any other application that I can use to send implicit data to OpENer? I don't care if I have to write the bits by hand. I couldn't find anything to do that in Molex or EtherNetIPTool. I just need something... 😭

mchiasson, 2019-05-23:
> _One tool I know is the EtherNet/IP Explorer which is a Scanner (master) and is written in C#._
YES! it works! This tool works! I can make an implicit connection that refreshes at the rate I want. Excellent! Thank you for this suggestion @CapXilinx

mchiasson, 2019-05-23:
I ended up being able to use `odva_ethernetip` after finally understanding what @CapXilinx told me the other day. I thought assembly data was only accessible via implicit messages, but you can get/set them via explicit class 0x04 attribute 3 while using the assembly ID as the instance ID. That will do perfectly fine for now, to be able to implement my _DEVICE_ until we receive our PLC.  Thank you @CapXilinx and @wpmccormick for all of your help!

closed 2019-05-23 by mchiasson