EIPStackGroup/OpENer #197: How to use EtherNet/IP Tool to the sync I/O Module in OpENer?

Farewellly, 2019-05-30:
I am a new to EtherNet/IP.Now, i use the etherNet/IP Tool to connect the OpENer in Linux. I am using (Process Data) function but it doesn't works.The CIP status show (INVALID SEGMENT IN CONNECTION PATH).
like this:
[image]
who can give me a hand to figure it out?

CapXilinx, 2019-06-03:
Hi,
as the error indicates your connection path is wrong. IO connection paths are non-trivial so it took me some time to get one to work (at the end I copied one from the CT tool)

So a correct IO connection path looks like this
200424972C962C64
where
2004 is the assembly class code
2497 - the configuration instance
2C96 - O->T connection point
2C64 - T->O connection point

Best,
Martin

CapXilinx, 2019-06-26:
As there was no further activity here, I close this issue

closed 2019-06-26 by CapXilinx