EIPStackGroup/OpENer #198: arduino support?

yanzixiang, 2019-06-03:
Is there any plan to support arduino ?

CapXilinx, 2019-06-03:
No, there is no plan for that, but we are working on a STM32 FreeRTOS version.

Best,
Martin

closed 2019-06-03 by CapXilinx

yanzixiang, 2019-06-03:
> No, there is no plan for that, but we are working on a STM32 FreeRTOS version.
> 
> Best,
> Martin

It is NOT publised ?

CapXilinx, 2019-06-03:
Yes, as we are currently working on the base setup and nothing is working right now.

yanzixiang, 2019-06-03:
https://github.com/espressif/arduino-esp32

recommand this,
this is much more cheaper and easier to use.

https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPServer/AsyncUDPServer.ino

crNewton, 2023-02-15:
> Yes, as we are currently working on the base setup and nothing is working right now.

I assume MCU support has been abandoned? 
Can I find the effort that has been done toward the STM32 FreeRTOS (or any other MCU) anywhere?