EIPStackGroup/OpENer #199: C++ SUPPORT ?

yanzixiang, 2019-06-04:
Is there any relationship between this and 
https://github.com/liftoff-sr/CIPster

Why not add some test like this to support C++
```C
#ifdef __cplusplus
extern "C" {
#endif

#ifdef __cplusplus
}
#endif
```

CapXilinx, 2019-06-05:
CIPster is a fork started about 2 years ago, changing OpENer to C++.

For your suggestion, this seems like a idea to me. I'll put it on my todo list.

wpmccormick, 2019-06-05:
I looked at that project for my C++ project as well and decided it would be better to just use shared libraries. OpENer now has shared library support which I am using to compile link C++ applications.

yanzixiang, 2019-06-06:
> I looked at that project for my C++ project as well and decided it would be better to just use shared libraries. OpENer now has shared library support which I am using to link C++ applications.

shared library support ?
how to use this ?
now i add 
```c
#ifdef __cplusplus
extern "C" {
#endif
```
at the start of all of the .h file 
and
```c
#ifdef __cplusplus
}
#endif
```
at the end of all of the .h file

CapXilinx, 2019-06-26:
for using OpENer as a shared library, you compile it with the OPENER_BUILD_SHARED_LIBS switch enabled. Please take a look on the readme file for details.

closed 2019-06-26 by CapXilinx