EIPStackGroup/OpENer #200: CipConnectionObjectCreate NOT implemented

yanzixiang, 2019-06-05:
I tried to connect my esp32 to an schneider tm241 PLC,
and use the default eds to config a network,
result in exception at DoublyLinkedListInsertAtHead()

```c
void AddNewActiveConnection(const CipConnectionObject *const connection_object)
{
  DoublyLinkedListInsertAtHead(&connection_list, connection_object);
  ConnectionObjectSetState(connection_object,
                           kConnectionObjectStateEstablished);
}
```

```c
void DoublyLinkedListInsertAtHead(DoublyLinkedList *const list,
                                  void *data) {
  OPENER_ASSERT(list->allocator != NULL)
  DoublyLinkedListNode * new_node = DoublyLinkedListNodeCreate(data,
                                                               list->allocator);
  if(NULL == list->first) {
    list->first = new_node;
    list->last = new_node;
  } else {
    new_node->next = list->first;
    list->first->previous = new_node;
    list->first = new_node;
  }
}
```

I think the error is caused by this,
can someone give a hand to implement this ?
at [cip/cipconnectionobject.c#L106](https://github.com/EIPStackGroup/OpENer/blob/8b8f249063ac09ce383caa9d7536dd7bcd0a3145/source/src/cip/cipconnectionobject.c#L106)
```C
CipConnectionObject *CipConnectionObjectCreate(const CipOctet *message) {
  assert(false); /* NOT IMLEMENTED */
  return NULL;
}
```

CapXilinx, 2019-06-05:
DoublyLinkedListInsertAtHead has no connection to CipConnectionObjectCreate.

I only added CipConnectionObjectCreate if dynamic Connection Object creation would be implemented in the future. Currently, it is not used by any means of the default application.

yanzixiang, 2019-06-06:
Yes,you are right.

closed 2019-06-06 by yanzixiang