EIPStackGroup/OpENer #201: Need help porting OpENer

ccvelandres, 2019-06-07:
I'm currently porting OpENer v2.3 to a STM32F3 FreeRTOS project. I'm quite confused what should be changed for the platform specific networkhandler. Is the generic_networkhandler meant to be an example? I need some help understanding how the ethernet encapsulation layer works. 

From the generated documentation from v2.3, I know that these have to be supplied, but i have no idea what the network handler works.
```c
// struct sockaddr_in
// AF_INET
// INADDR_ANY
// htons
// ntohl
// inet_addr

EipStatus AfterAssemblyDataReceived(CipInstance *instance);
EipStatus ApplicationInitialization();
EipBool8 BeforeAssemblyDataSend(CipInstance *pa_pstInstance);
void CheckIoConnectionEvent(unsigned int pa_unOutputAssembly, unsigned int pa_unInputAssembly, IoConnectionEvent pa_eIOConnectionEvent);
void *CipCalloc(unsigned pa_nNumberOfElements, unsigned pa_nSizeOfElement);
void CipFree(void *pa_poData);
void CloseSocket (const int socket_handle);
int CreateUdpSocket(UdpCommuncationDirection communication_direction, struct sockaddr_in *socket_data, CipUsint qos_for_socket);
void HandleApplication(void);
EipStatus ResetDevice(void);
EipStatus ResetDeviceToInitialConfiguration(void);
void RunIdleChanged (EipUint32 run_idle_value);
void RunIdleChanged(EipUint32 pa_nRunIdleValue);
EipStatus SendUdpData(struct sockaddr_in *socket_data, int socket_handle, EipUint8 *data, EipUint16 data_length);
```

Thanks,
Cedric

yanzixiang, 2019-06-08:
Do you use lwip ?

ccvelandres, 2019-06-08:
No. I'm using FreeRTOS+TCP

CapXilinx, 2019-06-08:
Hi,

for porting I suggest copying a port folder with the sample application without modifications.
Regarding the network handler, the generic network handler implements the hardware/system independent parts of the network handler system. In each port folder, there is the system dependent network handler part, where you have to supply the necessary functionality. 

So, these functions and define has to be implemented for FreeRTOS

#define OPENER_SOCKET_WOULD_BLOCK
EipStatus NetworkHandlerInitializePlatform(void);
void CloseSocketPlatform(int socket_handle);
int SetSocketToNonBlocking(int socket_handle);
MicroSeconds GetMicroSeconds(void);
MilliSeconds GetMilliSeconds(void);
int SetQosOnSocket(const int socket, CipUsint qos_value);

ccvelandres, 2019-06-10:
Got it. I browsed the other source files and found out that some have includes for "generic_networkhandler" and some have includes for "networkhandler". Should I name the file for "system dependent network handler" as "generic_networkhandler"? Or should i change the include directives to "networkhandler"?

I'm still quite confused with the distinction between the generic_networkhandler and the plain networkhandler.

CapXilinx, 2019-06-26:
@ccvelandres I tried to factor out platform independent functions needed by the OpENer into the generic networkhandler so this can be shared for all ports. Port-specific functions are in the networkhandler files in the corresponding ports folder. They are all named "networkhandler" so the correct one is pulled in, according to the selected CMake config.

ccvelandres, 2019-07-03:
I see. then I might end up doing making a new network handler for FreeRTOS since FreeRTOS already takes care of network implementation found on the generic_network handler.

The functions "HandleReceivedExplictUdpData" and "HandleReceivedExplictTcpData" is entry point for incoming packets and returns right? Does these function accept the UDP packets or just the payload?

CapXilinx, 2019-07-03:
Yes these functions are the entry points for handling explicit messages either received via UDP or TCP. It accepts the payload (received via recvfrom).

I am not sure what you mean by " FreeRTOS already takes care of network implementation found on the generic_network handler.", because the generic_networkhandler does not implement basic network operations, but gives the basic structure on handling ENIP messages.

ccvelandres, 2019-07-04:
> I am not sure what you mean by " FreeRTOS already takes care of network implementation found on the generic_network handler.", because the generic_networkhandler does not implement basic network operations, but gives the basic structure on handling ENIP messages.

My bad, I was wrong. I should have said I need to "restructure the code" or "replace some functions" to fit for FreeRTOS. e.g. bind -> FreeRTOS_bind 
So basically, I need to replace the winsock functions for the FreeRTOS functions.

I analyzed the code better this time. My understanding with the generic_networkhandler is that the function NetworkHandlerProcessOnce must be called regularly to check if there is pending packet to be received or sent, right? and then, ManageEncapsulationMessages() manages the implicit messaging right?

CapXilinx, 2019-07-10:
You are correct both times

CapXilinx, 2019-07-20:
@ccvelandres Everything clear now? Can we close this issue?

ccvelandres, 2019-07-22:
Clear for now, closing the issue for now

closed 2019-07-22 by ccvelandres