EIPStackGroup/OpENer #211: Testing with Ethernet/IP Explorer

jeremypatrickdahan, 2019-07-21:
First, thank you @CapXilinx for creating and maintaining this great library.

Though implicit messaging works fine for me, I am having trouble setting up an implicit connection to OpENer on my Windows 10 machines.
Being new to Ethernet/IP, I tried various scanners (Ethernet/IP Explorer, a simulated Fanuc controller, and [EEIP Java](https://github.com/rossmann-engineering/EEIP.Java/blob/master/Main.java)) and setups (on the same machine, on two machines connected via Ethernet, with Firewall on and off).
I am running the stock example app compiled with mingw, with all traces.
Below is one example of the kind of traces I get (always stopping at `connection manager: connect failed`).

@mchiasson, you seemed to have found the correct configuration to initiate a connection using Ethernet/IP Explorer (issue #195 ). Would you be kind enough to elaborate on how you did it ? I have tried for a few days with no success...

I was planning on contributing python bindings to simplify things for other beginners.

```
networkhandler: closing socket done 564
encap.c: Removing session
encap.c: Session removed
networkhandler: unsolicited UDP message on EIP unicast socket
Data received on UDP unicast:
List Identity
UDP unicast reply sent:
networkhandler: unsolicited UDP message on EIP global broadcast socket
Data received on global broadcast UDP:
List Identity
networkhandler: new TCP connection
>>> network handler: accepting new TCP socket: 568
New highest socket: 568
networkhandler: opened new TCP connection on fd 568
Entering HandleDataOnTcpSocket for socket: 568
Data received on tcp:
Handles data for TCP socket: 568
List services
TCP reply sent:
TCP response was not fully sent
Entering HandleDataOnTcpSocket for socket: 568
Data received on tcp:
Handles data for TCP socket: 568
Register session
Adds socket 568 to socket timers
Sets time stamp for socket 568
Sets time stamp for socket 568
TCP reply sent:
Sets time stamp for socket 568
TCP response was not fully sent
Entering HandleDataOnTcpSocket for socket: 568
Data received on tcp:
Handles data for TCP socket: 568
Send Request/Reply Data
Sockaddr type id: 8001
NotifyMessageRouter: routing unconnected message
NotifyMessageRouter: calling notify function of class 'connection manager'
notify: found instance 1
notify: calling ForwardOpen service
ForwardOpen: ConConnID 0, ProdConnID 1812642943, ConnSerNo 16991
We have a Non-Null request
We have a Non-Matching request
key: ven ID 1, dev type 12, prod code 65001, major 0, minor 0
classid 4 (assembly)
Configuration instance id 1
assembly: type bidirectional
connection point 150
connection point 100
Resulting PIT value: 256
IO Exclusive Owner connection requested
No PIT segment available
connection manager: connect failed
AssembleForwardOpenResponse: sending error response
notifyMR: notify function of class 'connection manager' returned a reply
Sets time stamp for socket 568
TCP reply sent:
Sets time stamp for socket 568
Entering HandleDataOnTcpSocket for socket: 568
networkhandler: socket: 568 - connection closed by client: 0 - The operation completed successfully.
````

CapXilinx, 2019-07-30:
Hi.

I just give it a shot, and by adding instances 100,150, and 151 (I also added 154 but did not use it), and issuing an implict message connection with 151 as config, 150 as output, and 100 as input, it worked for me with the sample application config.

Best
Martin

CapXilinx, 2019-08-05:
Hi,

is this issue resolved for you now, or do you need further help?

Best
Martin

jeremypatrickdahan, 2019-08-06:
Hi,
I am sorry for my late answer (I was fighting with a Fanuc controller...).
I tried again and it seems I have a compilation issue. I still cannot open an implicit connection but when using the DLL provided [here](https://github.com/mmasenlle/samples/tree/master/enip) in issue #136, I manage to open an implicit connection, but it quickly times out.
I think I am going to try on a Linux machine, being very unfamiliar with Windows compilation environments (I did follow all the steps of the readme and I had no compilation errors though).
Are there official pre-compiled binaries available for Windows and common Linux distros ?

Thank you very much for your help,

Best,

Jeremy Dahan

CapXilinx, 2019-08-07:
Hi,

unfortunatly, no there are no precompiled binaries available right now. But I also don't see how a compiler error could result in a timeout. What is your CMake config and are you running the EtherNet/IP explorer on a different machine or VM, or on the same machine as OpENer is running?

Best regards,
Martin

jeremypatrickdahan, 2019-08-08:
I did not use a VM and I did try running the Ethernet/IP Explorer on the same machine and on a different machine. Implicit messaging worked fine, but I still cannot establish an implicit connection.
I am ashamed to admit that I must focus on a different project and I cannot dive deeper into this.
I am very grateful for your help and time, and I will be sure to post a comment here when I get this solved, for reference.
Until then, I prefer to close the issue.
Thank you again

Best, 

Jeremy Dahan

closed 2019-08-08 by jeremypatrickdahan