EIPStackGroup/OpENer #213: OpENer stack prints bogus warning messages in trace mode

stm-at-esd, 2019-07-29:
Hi,

I' ve encountered the fact that the OpENer stack prints bogus warning messages in trace mode. The bogus message comes from HandleDataOnTcpSocket() in generic_networkhandler.c around line 786. 

The issue is that the code assumes that HandleReceivedExplictTcpData() returns the number of bytes to send putting it into the variable _number_of_bytes_to_send_ but in fact HandleReceivedExplictTcpData() returns an _EipStatus_ value (in the majority of cases!).

_number_of_bytes_to_send_ contains the _EipStatus_ value kEipStatusOkSend (i. e. 1) that triggers the send correctly but can't be used to compare if the TCP data has been sent completely which triggers the warning message. For the send() call on the other hand the _outgoing_message.used_message_length_ was used as needed!

This attached patch fixes the bogus warning message and renames _number_of_bytes_to_send_ to _need_to_send_ to make the name more appropriate.

[Fix_bogus_TCP_response_not_sent.patch.gz](https://github.com/EIPStackGroup/OpENer/files/3442250/Fix_bogus_TCP_response_not_sent.patch.gz)

I think the HandleReceivedExplictTcpData() (and HandleReceivedExplictUdpData() also) need a rework to consistently return an _EipStatus_ return type.

For HandleReceivedExplictTcpData() calling HandleReceivedSendRequestResponseDataCommand() or HandleReceivedSendUnitDataCommand() which in turn call NotifyConnectedCommonPacketFormat() that is returning the message size. I think at least these Handle*() should be converted to return _EipStatus_ because the real message size is conveyed using _outgoing_message_.

HandleReceivedInvalidCommand() having itself a return type of _EipStatus_ does return _outgoing_message.used_message_length_ which is wrong in my opinion.

Best regards,
    Stefan

CapXilinx, 2019-07-29:
Hi Stefan,

thanks for the fix. I put your suggestions for the Handle* functions into a separate issue, to better track the different topics. I think you are right and these functions should be streamlined. Some things you mentioned were introduced when I fixed the behavior when messages have to be repeated, so some code cleanup is surely required.

closed 2019-07-29 by CapXilinx

stm-at-esd, 2019-07-29:
Thank you for adding this so fast.

commit 7361b2be (2019-07-29): Closes #213 adds provided fix

Changes warning text when less data has been sent