EIPStackGroup/OpENer #215: Some files checkout with CRLF line endings under Linux

stm-at-esd, 2019-07-30:
When fixing a bug in ./source/src/enet_encap/encap.c I stumbled over the fact that a part of the C source files are checked out with CRLF or even mixed line endings under Unix / Linux. This makes it difficult to provide a proper patch to fix that bug.

The GitHub way to handle this is described here: https://help.github.com/en/articles/configuring-git-to-handle-line-endings

It would be nice if you could change the settings accordingly. The affected files are these:
`./contrib/msinttypes/inttypes.h:                          ASCII C++ program text, with CRLF line terminators
./src/enet_encap/encap.h:                                 ASCII C program text, with CRLF line terminators
./src/enet_encap/encap.c:                                 ASCII C program text, with CRLF line terminators
./src/cip/cipassembly.c:                                  ASCII C program text, with CRLF line terminators
./src/cip/cipioconnection.c:                              ASCII C program text, with CRLF line terminators
./src/cip/cipclass3connection.h:                          ASCII C program text, with CRLF line terminators
./src/cip/appcontype.h:                                   ASCII C program text, with CRLF line terminators
./src/cip/cipclass3connection.c:                          ASCII C program text, with CRLF line terminators
./src/cip/cipioconnection.h:                              ASCII C program text, with CRLF line terminators
./src/cip/appcontype.c:                                   ASCII C program text, with CRLF line terminators
./src/ports/WIN32/networkhandler.c:                       ASCII C program text, with CRLF line terminators
./src/ports/WIN32/sample_application/opener_user_conf.h:  ASCII English text, with CRLF line terminators
./src/ports/POSIX/networkhandler.c:                       ASCII C program text, with CRLF, CR line terminators
./src/ports/POSIX/sample_application/opener_user_conf.h:  ASCII English text, with CRLF line terminators
`

It would be nice if you could fix that state in the repository.
A raw patch for the bug in encap.c will follow in another issue.

CapXilinx, 2019-07-30:
Oh, wow. Thanks for the hint. That's strange as I already configured Git on Windows to check in Unix file endings.
I will do this immediately.

CapXilinx, 2019-07-30:
Hi, I hope this should be fixed now.

closed 2019-07-30 by CapXilinx

stm-at-esd, 2019-07-30:
I'll see if I can rebase my local git repository to the current head to have the files with consistent
line endings.

Best regards,
    Stefan

stm-at-esd, 2019-07-30:
A complete checkout with git clone is fine now on my development machine.

commit 091e2de3 (2019-07-30): Closes #215 Normalize all line endings