EIPStackGroup/OpENer #218: Some minor patches to fix typos and some code cleanup in posix/main.c

stm-at-esd, 2019-08-02:
Hi,

I would like to drop off some small patches regarding typos, some change of TRACE output
and a kind of code cleanup in the posix/main.c source. Besides the change of TRACE output
there is no functional change intended.

[patches_190802.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3461735/patches_190802.tar.gz)

The archive contains the patches created by _git format-patch_ and they could be integrated using _git am_.

I know it would be better to clone the OpENer repository on GitHub and send a pull request but
I need to talk to our GIT administrators before doing this about how to integrate that correctly in
our internal setup.

Best regards,
    Stefan

CapXilinx, 2019-08-02:
Hi,

thanks again.
I appreciate your patches regardless if you send them in form of patch files or pull requests. Code clean up has suffered in the last time, so every help here is appreciated. 

Best regards,
Martin

CapXilinx, 2019-08-07:
Tested and added to master branch

Thanks again!
Best regards,
Martin

closed 2019-08-07 by CapXilinx