EIPStackGroup/OpENer #219: Two patches to rework line ending normalisation and remove dead code from ciptcpipinterface.c

stm-at-esd, 2019-08-09:
Hi,

here are two small patches attached in the following archive. Can be applied with git am.
[patches_190809.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3487270/patches_190809.tar.gz)

1) This changes .gitattributes again to make the *.stc and *.eds files always being checked out
with CRLF line endings because they're used always under Windows.

2)  This patch removes some dead code that crept in by copy'n'paste from GetAttributeSingle() in the ciptcpipinterface.c code for GetAttributeSingleTcpIpInterface().

I think the function GetAttributeSingleTcpIpInterface() would need some more rework. To avoid code duplication I would call GetAttributeSingle() from that function for all attributes except for attribute #9 which would still be handled specially in GetAttributeSingleTcpIpInterface().

If you think that would be a good idea please tell me and I'll provide a patch for that.

I stumbled over these things because I need to extend the OpENer library for my project in the near future. Where could we discuss these possible extensions and how they could fit into the existing code? The Google group [here](https://groups.google.com/forum/#!forum/eip-stack-group-opener-developers) seems to be broken at the moment? At least a colleague of mine didn't get access to the group's postings with his Google account. There seemed to be nothing there.

Should I instead send you an email directly?

CapXilinx, 2019-08-12:
Hi,

regarding the Google Group, that's strange, as it usually works. I also don't see any request for joining the group. If the group really does not work you can either email me directly or use the Gitter channel.

For the patches, I will take a look asap. I will check the GetAttributeSingleTcpIpInterface, why it is specially handled right now. Could be a simple oversight, I changed several things with the GetAttribute-stuff in the past.

Best regards,
Martin

mmprestine, 2019-08-12:
The google group is working fine for me, I received all activity that has
been posted.

What fork are you planning?

Matt

CapXilinx, 2019-08-14:
Added the patches with commit 58bd0b9e913034772dd862c4b7eeae36cbee1c34

CapXilinx, 2019-08-14:
> I think the function GetAttributeSingleTcpIpInterface() would need some more rework. To avoid code duplication I would call GetAttributeSingle() from that function for all attributes except for attribute #9 which would still be handled specially in GetAttributeSingleTcpIpInterface

Oh yeah, now I remember. This function was the prototype for the new system for masking which attributes will be sent at an GetAttributeAll. When it worked we rolled it out to all objects, but it seems we forgot to update the prototype itself accordingly.
If you could supply a patch for this, I would be grateful.

Best regards,
Martin

stm-at-esd, 2019-08-15:
Hi,

I now have a patch for the rework of GetAttributeSingleTcpIpInterface() ready. This is the archive 
with the patches:
[patches_190815.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3505905/patches_190815.tar.gz)

It contains two patches (can be applied by git am):
1) Some code cleanup that avoids two warnings in TRACE mode for the POSIX variant of the code that are displayed when the network initialization code uses CloseSocket() to close internally used sockets. These are not connected and CloseSocket() calls CloseSocketPlatform() which tries a shutdown() on these sockets which triggers the error message in turn. I feel it is better to use a plain close() on the sockets in the platform specific network initialization code. That is what this patch introduces.
Similar (bogus) error messages are triggered by CloseSocket() being called from NetworkHandlerFinish() for the UDP listener sockets. But I have no quick idea about how to fix this
easily. Therefore atm. it stays as it is.

2) This is the rework of GetAttributeSingleTcpIpInterface(). The aim of the rework was to call GetAttributeSingle() for all attributes except attribute 9. Also the resulting manual handling of
attribute 9 should look very similar to the code in GetAttributeSingle() to make code reading
easier.

If you're happy with this rework this issue can be closed.

Best regards,
    Stefan

CapXilinx, 2019-08-26:
Thanks!
Merged in 8727e44cb0464e573841556d2ebbb8b0ba91e1e6

closed 2019-08-26 by CapXilinx