EIPStackGroup/OpENer #220: Multicast IO connections may fail on adapter devices with multiple network interfaces

stm-at-esd, 2019-08-12:
Hi,

when testing my device I had some problems to establish IO connections if they were configured as multicast connections. After some investigation I found out that me having a device with multiple network interfaces the multicast packets came out of the wrong network interface. 

I now direct the packets to the right interface by using a socket option to define the network
interface to be used for multicast packets originated from that socket.

[patches_190812.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3493962/patches_190812.tar.gz)

Three patches are included:
1) Small code reformatting and avoiding a compiler warning.
2) The patch for function CreateUdpSocket() to fix the outgoing network interface to be the one that has "our" EtherNet/IP IP adress.
3) Fix the function CreateUdpSocket() to return an error code in the case of a failure and NOT return with the socket number of the just created socket.

Regarding the patch number three there are still some questions / issues open and I would amend the patch if you say so:
a) On line 843 the function SetQosOnSocket() is called. If it returns an error only an error message is printed in TRACE mode but CreateUdpSocket() returns happily with success. Shouldn't a "CloseSocket(new_socket)" and a "return kEipInvalidSocket" be there after the printing of the error message?
b) On line 883 there is a setsockopt(..., SO_REUSEADDR,...) whose return code is NOT checked. This is in contrast to the same call on line 858 whose return code is checked an leads to an error return on failure of setsockopt(). Shouldn't the return code of the call on line 883 be checked in the same way? If you say that should be done I would amend this last patch and merge both calls to a single call in front of the "if (...)" construct on line 856.

Please  share your opinion and I'll change the patch number three as needed.

Best regards,
    Stefan

CapXilinx, 2019-08-14:
Hi,

the thing with multiple interfaces was known to me. I thought that I put this also into the documentation, but seems like I forgot. Until now I solved this by adding routes to the right interfaces, but I always wanted to add this to OpENer. So I welcome this change.

For a) I think the ENIP spec states, that errors in setting the QoS should not result in faults, but I have to recheck that.

For b) yes, this should be checked.

Thanks and best regards,
Martin

stm-at-esd, 2019-08-19:
Hi,

the patch number 3 from the patch archive patches_190812.tar.gz this issue report starts with has been changed to check also the return code of setsockopt(..., SO_REUSEADDR,...) as mentioned as case b) in the question section of my report.

Now the two setsockopt(..., SO_REUSEADDR,...) calls have been merged into a single call in the
code's main stream of the function whose return code is checked.

[patches_190812-b.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3516058/patches_190812-b.tar.gz)

This patch with the number 0003-... replaces the patch from the patches_190812.tar.gz archive.

The possible QoS errors from a) are still unchecked but I think it could stay as is. Perhaps you may add a comment in the source to explain why the error checking is missing and if it is really intended.

Best regards,
    Stefan

CapXilinx, 2019-08-26:
Thanks!
Merged in 8727e44cb0464e573841556d2ebbb8b0ba91e1e6

closed 2019-08-26 by CapXilinx