EIPStackGroup/OpENer #221: The POSIX and MINGW ports don't apply DSCP values for QoS to the IP TOS byte correctly

stm-at-esd, 2019-08-16:
Hi,

I stumbled today over the fact that the DSCP values for QoS were not correctly inserted in the TOS byte of the IP packets.  According to the EtherNet/IP specification they need to be shifted up into the DSCP field of the TOS byte (verified using wireshark). See the comments in the supplied patch below:
[patches_190816.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3510887/patches_190816.tar.gz)

Even if the setsockopt() call and value is fixed for MINGW too, it is supposed not to work as Windows handles QoS independent from socket settings. See [here for IP_TOS](https://docs.microsoft.com/en-us/windows/win32/winsock/ipproto-ip-socket-options) and [here](https://support.microsoft.com/en-us/help/248611/setsockopt-is-unable-to-mark-the-internet-protocol-type-of-service-bit) for more of this topic.

Best regards,
    Stefan

stm-at-esd, 2019-08-23:
Hi,

after looking on yet another Wireshark log while developing I found out that on some packets the recommended DSCP explict level markings were missing. This was true for the unconnected responses on requests received via the UDP listener sockets and the TCP SYN/ACK handshake to
initialize a TCP connection. To fix these issues I add another patch to this bug report.

[patches_190823.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3534292/patches_190823.tar.gz)

To quote its commit message here for information purposes:
> Fix to make QoS DSCP levels appear on all outgoing packets
    
    Responses for unconnected requests coming in via the UDP unicast or
    broadcast sockets did not contain the DSCP explicit level marking in the
    TOS field. To fix this call _SetQosOnSocket()_ for both UDP listener sockets
    which are also used for sending the corresponding responses.
    
    The TCP handshake packets (ACK, SYN+ACK) for incoming explicit TCP sessions
    also didn't contain the DSCP explicit level marking. To fix this call
    _SetQosOnSocket()_ for the TCP listen socket which makes the TCP handshake
    packets already contain the appropriate DSCP marking.
    
    On the other hand the _SetQosOnSocket()_ call on the newly created (by
    accept()) TCP socket can be omitted because that new socket inherits its
    settings from the listen socket (including the TOS setup).

This patch only changes _generic_networkhandler.c_ and should apply cleanly even if have some 
non public changes (more debug output) in my _generic_networkhandler.c_ here.

Best regards,
    Stefan

CapXilinx, 2019-08-26:
Thanks! Merged in 7800ca09eb12c2e5358f5915b7e9c76fc0c473a3

closed 2019-08-26 by CapXilinx