EIPStackGroup/OpENer #222: Optional AfterAssemblyDataSend and BeforeAssemblyDataReceived callbacks

ccvelandres, 2019-08-19:
Small enhancement i made for multi-threaded applications for OpENer. Since, I'm using OpENer on an embedded device, I need to save what little RAM I have. For my assembly data, I directly have OpENer read from a variable which can be accessed by multiple threads. So I need to protect each read/write access to prevent simultaneous read and write operations. 

Simple use case for suggested callbacks:
BeforeAssemblyDataSend (already exists) -> lock the variable for read operation
AfterAssemblyDataSend (proposed) -> unlock the variable after read operation

Here is the patch file. 
[patch.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3514752/patch.tar.gz)

1. This adds the prototype to opener_api.h and weak definitions on cipcommon.c
2. This also adds calls to proposed functions before/after the calls for its pair.

stm-at-esd, 2019-08-19:
Hi,

I had a look at the patch. The patch also changes the function _GetAttributeSingleTcpIpInterface()_ and adds a call to _AfterAssemblyDataSend()_ to this function.
Unfortunately this way you added more dead code to _GetAttributeSingleTcpIpInterface()_. _AfterAssemblyDataSend()_ will never be called from that function because it is never called with a instance of class _kCipAssemblyClassCode_. It is a specific handler for TcpIpInterface objects.  I proposed already to remove the dead call to _BeforeAssemblyDataSend()_ from that function in a patch I added to issue #219. I think it would be the best thing  if you simply would drop the change in _GetAttributeSingleTcpIpInterface()_ from your patch.

I believe also that you should be more specific in the description of the functions. For instance the _BeforeAssemblyDataSend()_ / _AfterAssemblyDataSend()_ function pair have both a description of

> Within this function the user can update the data of the assembly object
 before it gets sent. The application can inform the application if data has
 changed.

I think that this is confusing. I remember it took me some time to understand when these callbacks 
are called and to what purpose.

Wouldn't it be better to have for _BeforeAssemblyDataSend()_ something like that:

> Within this function the user can update the data of the assembly object
 before it gets sent. The application can inform the OpENer stack if data 
 has changed.
In a multi-threaded environment the user may need to lock the assembly 
data here before it is sent.

And to have for _AfterAssemblyDataSend()_ something like that:

> This function is called after the assembly data are sent. The user needs to
implement this function only in a multi-threaded environment to free a lock 
that may have been acquired in the _BeforeAssemblyDataSend()_ callback.

I think that would clarify the usage much more. The same holds true for the
other function pair.

Best regards,
    Stefan

ccvelandres, 2019-08-20:
Hi, 

I just saw the issue you mentioned. I'll change the patch to reflect your comments. Same for the function descriptions. I'll try to finish making the patch this week once I have time.

Regards, 
Cedric

CapXilinx, 2020-01-15:
HI. As there has been a patch by @stm-at-esd which includes this functionality, I close this issue as it seems resolved for me.

closed 2020-01-15 by CapXilinx