EIPStackGroup/OpENer #223: wrong device type ?

yanzixiang, 2019-08-19:
[image]

I want to use OpENer with an Omron PLC which support EIP,
and got this error,
I think there is some problem witch the default EDS,
but I don't know where should be corrected,
please help

stm-at-esd, 2019-08-19:
Hi,

looking at the EDS here on GitHub (current state) you have in the [Device] section a device type of "12" i. e. "Communications Adapter" and a revision of 2.3 (i. e. "MajRev = 2" / "MinRev = 3").

Try to send a "ListIdentity" command to your device and see what it responds. Or have a Wireshark log of the conversation between the Omron device and your device and look for "ListIdentity" responses.

On the other hand you should know what you've compiled in.

Make sure that the EDS file installed in the Omron programming IDE for the PLC matches the one you're looking at and your device.

Maybe that helps,
    Stefan

CapXilinx, 2019-08-20:
Hi,

i checked it right now with my compiled versio of OpENer, device type and version is exactly as @stm-at-esd wrote, and are the same as in the current EDS file.

Best way to check your settings is, as @stm-at-esd suggested, to send a ListIdentity request to check your OpENer. You can do this e.g., with the Molex ENIP tool.

Best regards,
Martin

CapXilinx, 2019-08-21:
I guess this is resolved?

yanzixiang, 2019-08-21:
yes,thanks.

closed 2019-08-21 by yanzixiang