EIPStackGroup/OpENer #224: What does PIT(ProductionInhibitTime) mean ?

yanzixiang, 2019-08-20:
20:17:21.254 -> [E][generic_networkhandler.c:612] HandleDataOnTcpSocket(): Entering HandleDataOnTcpSocket for socket: 57
20:17:21.254 -> [E][generic_networkhandler.c:742] HandleDataOnTcpSocket(): Data received on tcp:
20:17:21.254 -> [E][encap.c:158] HandleReceivedExplictTcpData(): Handles data for TCP socket: 57
20:17:21.254 -> [E][encap.c:214] HandleReceivedExplictTcpData(): Send Request/Reply Data
20:17:21.254 -> [E][cipmessagerouter.c:182] NotifyMessageRouter(): NotifyMessageRouter: routing unconnected message
20:17:21.322 -> [E][cipmessagerouter.c:218] NotifyMessageRouter(): NotifyMessageRouter: calling notify function of class 'connection manager'
20:17:21.322 -> [E][cipcommon.c:88] NotifyClass(): notify: found instance 1
20:17:21.322 -> [E][cipcommon.c:99] NotifyClass(): notify: calling ForwardOpen service
20:17:21.322 -> [E][cipconnectionobject.c:215] ConnectionObjectSetState(): ConnectionObjectSetState
20:17:21.322 -> [E][cipconnectionobject.c:157] ConnectionObjectInitializeFromMessage(): ForwardOpen: ConConnID 0, ProdConnID 0, ConnSerNo 271
20:17:21.322 -> [E][cipconnectionmanager.c:539] ForwardOpen(): We have a Non-Null request
20:17:21.356 -> [E][cipconnectionmanager.c:548] ForwardOpen(): We have a Non-Matching request
20:17:21.356 -> [E][cipconnectionmanager.c:1116] ParseConnectionPath(): key: ven ID 152, dev type 145, prod code 18, major 1, minor 1
20:17:21.356 -> [E][cipconnectionmanager.c:1174] ParseConnectionPath(): classid 4 (assembly)
20:17:21.356 -> 
20:17:21.356 -> [E][cipconnectionmanager.c:1189] ParseConnectionPath(): Configuration instance id 151
20:17:21.356 -> [E][cipconnectionmanager.c:1276] ParseConnectionPath(): assembly: type bidirectional
20:17:21.391 -> [E][cipconnectionmanager.c:1300] ParseConnectionPath(): connection point 150
20:17:21.391 -> [E][cipconnectionmanager.c:1300] ParseConnectionPath(): connection point 100
### 20:17:21.391 -> [E][cipconnectionmanager.c:1379] ParseConnectionPath(): Resulting PIT value: 256
20:17:21.391 -> [E][cipioconnection.c:280] EstablishIoConnection(): EstablishIoConnection
20:17:21.391 -> [E][appcontype.c:167] GetIoConnectionForConnectionData(): IO Exclusive Owner connection requested
### 20:17:21.391 -> [E][cipioconnection.c:82] ProcessProductionInhibitTime(): No PIT segment available
### 20:17:21.458 -> [E][cipconnectionmanager.c:442] HandleNonNullNonMatchingForwardOpenRequest(): connection manager: connect failed
### 20:17:21.458 -> [E][cipconnectionmanager.c:774] AssembleForwardOpenResponse(): AssembleForwardOpenResponse: sending error response
20:17:21.458 -> [E][cipconnectionobject.c:215] ConnectionObjectSetState(): ConnectionObjectSetState
20:17:21.458 -> [E][cipmessagerouter.c:237] NotifyMessageRouter(): notifyMR: notify function of class 'connection manager' returned a reply
20:17:21.458 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
20:17:21.458 -> [E][generic_networkhandler.c:780] HandleDataOnTcpSocket(): TCP reply sent:
20:17:21.458 -> [E][socket_timer.c:21] SocketTimerSetLastUpdate(): Sets time stamp for socket 57
20:17:21.458 -> [E][generic_networkhandler.c:405] NetworkHandlerProcessOnce(): Socket Loop done

azoitl, 2019-08-20:
Hi,

the PIT is a timer for application triggered or change of state connection the PIT defines the minimal time between to message productions.

BR,
Alois

CapXilinx, 2019-08-21:
The TODO just says, that this working code should be improved for maintainability and readability.
As the question has been answered I close this issue now.

closed 2019-08-21 by CapXilinx