EIPStackGroup/OpENer #225: How to Debug SetupIoConnectionOriginatorToTargetConnectionPoint ?

yanzixiang, 2019-08-20:
[image]

23:19:45.240 -> [E][cipioconnection.c:104] SetIoConnectionCallbacks(): start
23:19:45.240 -> [E][cipconnectionobject.c:817] ConnectionObjectGeneralConfiguration(): start
23:19:45.240 -> [E][cipconnectionobject.c:810] ConnectionObjectResetProductionInhibitTimer(): start
23:19:45.240 -> [E][cipconnectionobject.c:848] ConnectionObjectGeneralConfiguration(): end
23:19:45.240 -> [E][cipioconnection.c:320] EstablishIoConnection(): O->T: 2,T->O: 1
23:19:45.240 -> [E][cipioconnection.c:116] SetupIoConnectionOriginatorToTargetConnectionPoint(): start
23:19:45.273 -> [E][cipconnectionobject.c:371] ConnectionObjectGetTransportClassTriggerTransportClass(): start
23:19:45.273 -> [E][cipconnectionobject.c:398] ConnectionObjectGetTransportClassTriggerTransportClass(): transport_class_trigger : 1
23:19:45.273 -> [E][cipconnectionmanager.c:451] HandleNonNullNonMatchingForwardOpenRequest(): connection manager: connect failed

How can I Debug why the connection failed ?

CapXilinx, 2019-08-21:
You can't directly, as the timing is essential for the correct behavior of Originator and Target, that's why the traces are there. I also suggest recording a trace with Wireshark for more information. From that point on you have to draw your conclusions based on observation and code, what could have gone wrong.

BR,
Martin

yanzixiang, 2019-08-21:
[image]

Do you have some PDF that can help?
what are they doing when
### SetupIoConnectionOriginatorToTargetConnectionPoint
and
### SetupIoConnectionTargetToOriginatorConnectionPoint

CapXilinx, 2019-08-21:
No, there is no PDF available. Both function set up the connection endpoints of the Class1 connection and are internal to OpENer.
I don't know what model of Omron was used on the PlugFest, but I know that OpENer worked flawlessly with it.

Either your router has a port replication option or you could buy a smart hub capable of doing this, connecting it in between your PLC and your device.

yanzixiang, 2019-08-21:
@No, there is no PDF available. Both function set up the connection endpoints of the Class1 connection and are internal to OpENer.
Yes,I am digging hard int to OpENer.
I just can NOT find out which method got errer,
because ConnectionObjectGetTransportClassTriggerTransportClass just ends up without an error code.

@I don't know what model of Omron was used on the PlugFest, but I know that OpENer worked flawlessly with it.
What is PlugFest ? I have never hear about it.

@A router has a port replication option,
or a smart hub capable,
Can you give me some examples ?

CapXilinx, 2019-08-22:
if ConnectionObjectGetTransportClassTriggerTransportClass is returning an error (I guess you mean it returns kConnectionObjectTransportClassTriggerTransportClassInvalid), than your PLC is sending a wrong data package, because it is only extracting the corresponding bits from the FowardOpenRequest, and puts it into a named constant value.

PlugFests are activities of the ODVA, where ENIP devices are tested with advanced test tools, and to check interoperability between several vendors.

For a smart router please just google. It just needs the feature of port replication.

yanzixiang, 2019-08-22:
> if ConnectionObjectGetTransportClassTriggerTransportClass is returning an error (I guess you mean it returns kConnectionObjectTransportClassTriggerTransportClassInvalid), than your PLC is sending a [...]

23:19:45.273 -> [E][cipconnectionobject.c:398] ConnectionObjectGetTransportClassTriggerTransportClass(): transport_class_trigger : 1

CIP_CONNECTION_OBJECT_TRANSPORT_CLASS_TRIGGER_TRANSPORT_CLASS_1

I think this is a valid transport_class_trigger

CapXilinx, 2019-08-22:
Yes. Every value, except kConnectionObjectTransportClassTriggerTransportClassInvalid, indicates a valid value. Class1 is to be expected for IO connections with OpENer

yanzixiang, 2019-08-26:
[20190826_0001.zip](https://github.com/EIPStackGroup/OpENer/files/3541111/20190826_0001.zip)
[image]
[image]

please help to analyse the pcap file.

CapXilinx, 2019-08-26:
The problem is, that you only give two Connection Points (100 and 150), but the standard Exclusive Owner connection is configured to need three connection points for output, input, and config endpoints (151 for config). Perhaps this is a bug in OpENer, I couldn't find something on this in the spec if the config connection point must be present in this case or not.

CapXilinx, 2019-08-26:
Please try this branch, if it solves your problem https://github.com/EIPStackGroup/OpENer/tree/Issue_225

CapXilinx, 2019-08-26:
After checking with Pub 70 of the ODVA ("Recommended Functionality for EtherNet/IP Devices"), my opinion in this matter is, that OpENer is behaving correct by rejecting this connection. Your used device has to send 3 connection points (input, output, and configuration connection points).

CapXilinx, 2019-08-26:
Although, I am still not sure. According to Table 3-5.13, when no data segment is available, then no configuration point is given.

CapXilinx, 2019-08-27:
Its used for passing configuration data to the connection endpoint. It is part of the Forward Open request as a data segment. But for OpENer you do not need to actually send data (also valid ENIP), but you have to specify the coniguration assembly (151). This is all explained in the ENIP specification available at [ODVA](www.odva.org)
Have you tried the branch I provided for you?

I do not know this software. A better test would be to use the ODVA Conformance Test Tool

yanzixiang, 2019-08-27:
> After checking with Pub 70 of the ODVA ("Recommended Functionality for EtherNet/IP Devices"), my opinion in this matter is, that OpENer is behaving correct by rejecting this connection. Your used device has to send 3 connection points (input, output, and configuration connection points).
> Although, I am still not sure. According to Table 3-5.13, when no data segment is available, then no configuration point is given.

Can you explain what is the configuration connections point used for And how to send it ?
I can NOT find any option to define it in OMRON's program.

And and test my custom board with this [client library](http://www.eeip-library.de)

yanzixiang, 2019-08-27:
> After checking with Pub 70 of the ODVA ("Recommended Functionality for EtherNet/IP Devices"), my opinion in this matter is, that OpENer is behaving correct by rejecting this connection. Your used [...]

Can you explain what is the configuration connections point used for And how to send it ?
I can NOT find any option to define it in OMRON's program.

And and test my custom board with this [client library](http://www.eeip-library.de)

yanzixiang, 2019-08-28:
> Its used for passing configuration data to the connection endpoint. It is part of the Forward Open request as a data segment. But for OpENer you do not need to actually send data (also valid ENIP), [...]

Can you point to the PDF file,or just upload it ?

> Have you tried the branch I provided for you?

I have NOT, will try it later.
 
> I do not know this software. A better test would be to use the ODVA Conformance Test Tool

where can i get it ?

yanzixiang, 2019-08-28:
I Tested https://github.com/EIPStackGroup/OpENer/tree/Issue_225.
Nothing changed.

It's defined here
[kConnectionManagerExtendedStatusCodeErrorInvalidSegmentTypeInPath](https://github.com/EIPStackGroup/OpENer/blob/fadb3aa1dce82be832a9bad24c9c384f303ac550/source/src/cip/cipconnectionmanager.h#L116)
```c
kConnectionManagerExtendedStatusCodeErrorInvalidSegmentTypeInPath = 0x0315,
```

It is returned here,
But I don't know why.
[ParseConnectionPath](https://github.com/EIPStackGroup/OpENer/blob/f76f479b17c7a700d7f832612cc9064be091f005/source/src/cip/cipconnectionmanager.c#L1050)

yanzixiang, 2019-08-28:
[image]

I don't know why,
but the default value of the coniguration assembly is 1 if unspecified.
After change it to 1,
got another errorcode 
# kConnectionManagerExtendedStatusCodeErrorInvalidOToTConnectionSize

But they are all 32.

CapXilinx, 2019-08-29:
> Can you point to the PDF file,or just upload it ?
As already in my post, please take a look at the ODVA homepage.

> where can i get it ?
At ODVA. You have to become a member anyhow to use OpENer commercialy or to redistribute at, as in the license agreement.

Your change to 1 is clearly wrong. You just get another error message as you are cheating your way through this check.

I took another look on your trace and compared it to my tools. The requested assembly instance (1 in your case) is wrong. This represents the configuration connection point and needs to be 0x97 for OpENer. You can see this in the sample configuration of OpENer, and in some already answered issues.

I would recommend to you, that you first take a deeper look into ENIP before trying to port OpENer to a custom board.
You can get basic information form the ODVA publication download page.
Document Pub 213 -"Quick Start for Vendors" is a good starting point. Also, I would recommend to you to get in touch with ODVA and get yourself at least the CIP Specifications Vol. 1 and Vol. 2.

yanzixiang, 2019-08-29:
So there is NOT a way to just disable the configuration connection point check ?

CapXilinx, 2019-08-29:
The comparison to zero I added is disabling the check, if no configuration connection point is provided. If you send one in your Forward Open it has to match

yanzixiang, 2019-08-30:
# the default value of the coniguration assembly is 1 if unspecified.
After change it to 1,
```c
#define DEMO_APP_CONFIG_ASSEMBLY_NUM               1 //0x097
```
got another errorcode
# kConnectionManagerExtendedStatusCodeErrorInvalidOToTConnectionSize

CapXilinx, 2019-08-30:
as the error says your O->T connection size is wrong. If you send a data segment, it has to match die length of the config array. Otherwise your O->T message size is wrong

yanzixiang, 2019-08-30:
After 
```c
#define OPENER_CONSUMED_DATA_HAS_RUN_IDLE_HEADER
```
I got a success.

CapXilinx, 2019-08-30:
Great that it works for you now, but you don't define this option, but should activate it via CMake

closed 2019-08-30 by CapXilinx

CapXilinx, 2019-08-30:
@yanzixiang I would ask you to close your resolved issues yourself in the future.

yanzixiang, 2019-08-31:
> Great that it works for you now, but you don't define this option, but should activate it via CMake

I am using ardunio and I am NOT able to use CMake.

> @yanzixiang I would ask you to close your resolved issues yourself in the future.

OK.