EIPStackGroup/OpENer #226: Patch: Get rid of error messages triggered by CloseSocket() on  POSIX

stm-at-esd, 2019-08-26:
Hi,

I've recently fixed the error messages thrown when closing an IO connection on the POSIX platform. For each closed IO connection socket (UDP socket) an "Error Socket not connected" message was thrown. The same occurred when shutting down OpENer when the UDP listener sockets were closed.

Description of patch 0001:

The reason for this was that the platform dependent _CloseSocketPlatform()_ function called _shutdown()_ for the socket which throws an error on all OpENer UDP sockets because they're not connected.

To quote the commit message of the patch 0001:

> The POSIX _CloseSocketPlatform()_ low level function called _shutdown()_ which
only returns without error on TCP sockets. This triggered some bogus error
messages about "Could not close socket" with the POSIX port in normal
(error free) operation. I. e. for each normally closed IO connection.
> 
> The generic network handler code already utilized _CloseUdpSocket()_ and
_CloseTcpSocket()_ to treat UDP and TCP sockets unequally.
> 
> The fix is to move the _shutdown()_ code into a new _ShutdownSocketPlatform()_
function and call this function only from _CloseTcpSocket()_. The Windows
versions of that function are empty and only contain a compiler warning
that should be removed if the code checked under Windows.
>
> The function _CreateUdpSocket()_ now also uses the correct _CloseUdpSocket()_
instead of _CloseSocket()_. Fixed also a socket leak in this function that
in some error cases the created (failed) socket was not closed.

Under Windows the _CloseSocketPlatform()_ did not call _shutdown()_. Therefore the
implementation of _CloseSocketPlatform()_ for **WIN32** and **MINGW** is empty. 
I only added a compiler warning there to give the maintainer of the Windows variant a
hint that about this decision should be thought again. If the _shutdown()_ call is not
needed under Windows then the lines with the warnings can simply be removed before
the patch is applied.

Description of patch 0002:

This patch only has minor changes / cleanup only:

> - Typo in CipEthernetLinkObject's Doxygen comment.
> - Missing '\n' in trace output in CipIdentitySetExtendedDeviceStatus().
> - Remove superfluous _get_attribute_all_mask_ member of CipClass which
>   was superseded by the _get_all_bit_mask_member_.

I hope that these patches will be useful.

[patches_190823.tar.gz](https://github.com/EIPStackGroup/OpENer/files/3541494/patches_190823.tar.gz)

Best regards,
    Stefan

CapXilinx, 2019-08-26:
As far as I remember, the variant with shutdown was introduced, as an experiment for several strange errors with the conformance tester. Back then I had to change a lot of different behaviors in OpENer to get it to do as the CT expected.
It is easily possible, that the whole shutdown thing is not needed, but I never replaced it, as I had several other things to fix.

We could try a variant, which does not call shutdown at all.

Best regards,
Martin

stm-at-esd, 2019-08-26:
I think internally _shutdown()_ is called in any case (by close()). And I have no objections on calling _shutdown()_ on TCP sockets which sounds sensible to me making the connection tear-down more explicit.

Indeed the patch makes the _ShutdownSocketPlatform()_ appear before _RemoveSocketTimerFromList()_ which changes the timing a little bit. 

But in the meantime I could try and define the _ShutdownSocketPlatform()_ to become empty
and see if the conformance test will complain. I'm using CT 16 atm. and the test seems to work
so far (except some complaints about the response times for the broadcast ListIdentity request not being random enough).

CapXilinx, 2019-08-26:
I am also using CT-16 and I also got the complaint about randomness today, but only in one test run. I thought this was just a fluke. I also implemented an xor Shirt RNG, which should in theory perform better, than the Rand standard library function.

stm-at-esd, 2019-08-27:
Regarding the randomness problem I've had a look at my lab notices and I once did 10 encapsulation tests where 4 of them failed with the bad randomness complaint.

But the conformance test only displays (and uses for calculation) 10 response times of the ListIdentity response. From my point of view that are not enough probes to decide the quality of the random generator with a statistical significance.

So I decided to postpone this issue and ask the ODVA guys later for a good explanation or at least that they send more ListIdentity requests to achieve a better statistical significance.

On the other hand I did some simple tests to view the distribution of frequency for the algorithm used in OpENer. If using enough rand() calls the frequency is evenly distributed. Therefore I believe it is more problem of the check in CT-16 than a real problem of the implementation.

CapXilinx, 2019-08-27:
Thanks for checking! I did 10 to 15 runs yesterday with only one time with this error.

FYI - the original reason for the XOR shift was, that a platform I used did not come with a rand function in its standard library.

CapXilinx, 2019-08-27:
Added with commit f9968c5328a6238906841f7e7300ff5817ce894a

closed 2019-08-27 by CapXilinx