EIPStackGroup/OpENer #253: TCP response was not fully sent

yanzixiang, 2019-08-30:
```c
if (number_of_bytes_to_send > 0) {
      OPENER_TRACE_INFO("TCP reply sent:\n");

      data_sent = send(socket, (char *) outgoing_message.message_buffer,
                       outgoing_message.used_message_length, 0);
      SocketTimer *socket_timer = SocketTimerArrayGetSocketTimer(
        g_timestamps,
        OPENER_NUMBER_OF_SUPPORTED_SESSIONS,
        socket);
      SocketTimerSetLastUpdate(socket_timer, g_actual_time);
      if (data_sent != number_of_bytes_to_send) {
        OPENER_TRACE_WARN("TCP response was not fully sent data_sent: %d,number_of_bytes_to_send : %d \n",
		  data_sent,
		  number_of_bytes_to_send
		);
      }
    }
```

After a time for some minutes of connect link,
I got this WARN,
And CJ1W-EIP21 give a errer.
What does this mean ?
There is something wrong with my board or the OpENer stack ?

CapXilinx, 2019-08-30:
Based on the provided information, I cannot give you any hint.

stm-at-esd, 2019-08-30:
Hi,

could it be that you're using an "old" version of OpENer? Which git revision you're using?

But from my experience and the code snippet you sent (still `number_of_bytes_to_send` is there) it seems that you stumbled over the same problem than me. Please refer to issue #213 and the explanation.

In short: The warning message is printed in most cases even the send itself is ok.

Make sure that this patch is applied, retry and see if you have still the error.

yanzixiang, 2019-08-31:
@stm-at-esd  You are right.

closed 2019-08-31 by yanzixiang